greatscottgadgets/greatfet #9: Host utilities and libraries not installed

docprofsky, 2016-08-13:
Running `python setup.py install` will only copy the `.egg` into `site-packages/`, but will not unpack it, or install the host utilities anywhere.

I am running OS X 10.10.5 (Yosemite) with Python 2.7 installed with macports into `/opt/local`.
On OS X Python's  `site‑packages/` is located at `/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/`.

dominicgs, 2016-08-13:
Host tools are now installed as of commit 5d50df0

As we're using setuptools, Python should take care of the paths for us.  There is no need for the .egg file to be unpacked.

docprofsky, 2016-08-14:
I have tested it and confirm that 5d50df0 fixes this.

closed 2016-08-14 by docprofsky