greatscottgadgets/greatfet #92: USB1 host example

mossmann, 2018-07-06:
I would like an example implementation that can be called from Python (and maybe a command-line tool) on the host that demonstrates USB1 functioning in host mode.  For example, it could just read descriptors from a device attached to USB1 and pass those along to the host attached to USB0.

mossmann, 2018-07-10:
This should probably be implemented with https://github.com/ktemkin/Facedancer

ktemkin, 2018-09-05:
I've cleaned up and merged a good chunk of the host API into FaceDancer, which I think provides the cleanest example of using USB1 from the host:

```
from facedancer import FacedancerUSBHostApp

u = FacedancerUSBHostApp(verbose=5)
u.initialize_device()
print(u.get_device_descriptor())
```

This still needs a bit of testing, but it's there and should work. We've also been able to do this through the GlitchKit API since December:

```
# if we don't set up any triggering, this will only perform the control request itself
descriptor = gf.glitchkit.usb.capture_control_in(
             request= gf.glitchkit.usb.GET_DESCRIPTOR,
             value  = gf.glitchkit.usb.GET_DEVICE_DESCRIPTOR,
             length = 18
         )
```

Finally, it's also worth noting there's a nice API to do this directly in the firmware, if we want to add it to the future self-test API:

```
	// Read the device descriptor, which contains our VID and PID.
	usb_descriptor_device_t device_descriptor;
	usb_host_read_device_descriptor(&usb_peripherals[1], control_endpoint, &device_descriptor);

	// If our VID or PID don't match, fail out.
	if (device_descriptor.idVendor != SWITCH_RCM_VID)
		return EINVAL;
	if (device_descriptor.idProduct != SWITCH_RCM_PID)
		return EINVAL;
```

see e.g: https://github.com/Cease-and-DeSwitch/rcm-modchips/blob/master/greatfet-modchip/main.c#L134

ktemkin, 2018-09-05:
For e.g. factory testing of USB1, I'd suggest using the GlitchKit backend, as it has the least external dependencies:

``` python 
import greatfet

gf = greatfet.GreatFET()
descriptor = gf.glitchkit.usb.capture_control_in(
             request= gf.glitchkit.usb.GET_DESCRIPTOR,
             value  = gf.glitchkit.usb.GET_DEVICE_DESCRIPTOR,
             length = 18
         )
print(descriptor)
```

More general-purpose examples of using USB1 from the host and device are probably best found in FaceDancer (e.g. https://github.com/ktemkin/Facedancer/blob/master/facedancer-host-enumeration.py) or one of the downstream projects, like the Switch modchips (https://github.com/Cease-and-DeSwitch/rcm-modchips/blob/master/greatfet-modchip/main.c).

@mossmann: Are you looking for anything else in an example, or should we call this one closed?

mossmann, 2018-12-12:
Neither of these methods is working for me.  The FaceDancer method (after doing `s/vendor_request_/comms._vendor_request_/` on FaceDancer) hangs on u.get_device_descriptor() after "Issuing setup packet".  The GlitchKit method errors on gf.glitchkit.usb.capture_control_in() with:

```
Traceback (most recent call last):
  File "/tmp/usb1.py", line 10, in <module>
    length = 18
  File "/usr/lib/python3.6/site-packages/GreatFET-0.0-py3.6.egg/greatfet/glitchkit/usb.py", line 126, in capture_control_in
    raise e
  File "/usr/lib/python3.6/site-packages/GreatFET-0.0-py3.6.egg/greatfet/glitchkit/usb.py", line 120, in capture_control_in
    raw = self.board.comms._vendor_request_in(vendor_requests.GLITCHKIT_USB_RESULT_LENGTH, length=4, timeout=timeout)
  File "/usr/lib/python3.6/site-packages/pygreat-0.0-py3.6.egg/pygreat/comms_backends/usb.py", line 138, in _vendor_request_in
  File "/usr/lib/python3.6/site-packages/pygreat-0.0-py3.6.egg/pygreat/comms_backends/usb.py", line 126, in _vendor_request
  File "/usr/lib/python3.6/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/lib/python3.6/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
```

Both methods get far enough to enable USB1_VBUS.

ktemkin, 2018-12-12:
Looking into this now -- I have a branch that has GlitchKit ported to the new comms API; I'll check to see if this issue is also happening there.

By chance, can I ask what USB device you're testing that with?

dominicgs, 2018-12-12:
The other device is another GreatFET, using FaceDancer on USB1.

ktemkin, 2018-12-12:
There was another subtle alignment issue hiding in here. Fixed, and I brought in the code I had that moved GlitchKit to the new comms API while I was at it.

New code is in this PR: https://github.com/greatscottgadgets/greatfet/pull/182

I can verify that the following works for me:
```
import greatfet

gf = greatfet.GreatFET()
descriptor = gf.glitchkit.usb.capture_control_in(
             request= gf.glitchkit.usb.GET_DESCRIPTOR,
             value  = gf.glitchkit.usb.GET_DEVICE_DESCRIPTOR,
             length = 18
         )
print(descriptor)
```

@mossmann Can you validate that, with the PR'd code, this works for manufacturing test?

mossmann, 2018-12-13:
That works.  Thank you!

It does not fail gracefully, however.  Without a device connected to USB1, it times out, and the GreatFET is unresponsive.

Also, after success, USB1_VBUS remains on.  Is that the intended behavior?

ktemkin, 2018-12-14:
> It does not fail gracefully, however. Without a device connected to USB1, it times out, and the GreatFET is unresponsive.

Generally, these don't fail super-gracefully; the GlitchKit code is meant to have few error-handling paths in the timing-critical sections in order to give consistent timing where possible during fault injection.

This shouldn't hang; though -- I think there's a workaround in there now that's making things wait for device startup infinitely, which probably shouldn't be the case. I'll fix things a bit, but I don't have a USB-OTG cable with me and thus won't be able to test the resultant PR for a couple of days.

> Also, after success, USB1_VBUS remains on. Is that the intended behavior?

Yes -- often this is used for a bus-powered device, and with GlitchKit, often the interesting behavior continues for a while after the request is complete in a way that's visible to the CW side. I can see cases where you might want VBUS to turn off, though -- I'll tweak this so it's configurable.

ktemkin, 2018-12-19:
See: https://github.com/greatscottgadgets/greatfet/pull/193

closed 2018-12-20 by ktemkin