greatscottgadgets/greatfet #112: sigrok support for logic analysis

mossmann, 2018-07-10:
Make it easy to use GreatFET as a logic analyzer with sigrok.

dominicgs, 2018-07-19:
https://sigrok.org/gitweb/?p=libsigrok.git;a=blob;f=HACKING;hb=HEAD

ktemkin, 2019-08-20:
Moving this from "next release" to the release after, as this isn't a blocker for the release (it's independently versioned and instead depends on us getting the release out there).

In short, I don't want to submit anything to upstream Sigrok until we have a release already out for folks to test against.

antzware, 2021-02-01:
Hello. Do you already have something on this that you can share and it is not officially released? Otherwise I would like to have a try at this myself if you have an example of accessing the API on the board with libusb from C code.

ktemkin, 2021-02-01:
I believe @Qyriad has a (nearly completely done?) branch: https://github.com/Qyriad/libsigrok/tree/greatfet-driver

luizribeiro, 2021-08-02:
I checked out @Qyriad's and @ktemkin's [awesome] work today and got it to work after a few minor adjustments: https://github.com/luizribeiro/libsigrok/commit/a8048dac08ee5de6120e26272daaa04fc38ba599

For now it only supports 16 channels and you can only use one GreatFET at once (I believe, I couldn't test this). It also doesn't support any scan options on the driver. It does support triggers though! I spent a bit of time today playing with it and it seems to be working well.

I'm planning on turning it into a pull request soon - and we can add the remaining features as we go. @Qyriad, I see we only have @ktemkin's name on the Copyright headers. How would you like for me to attribute your work?

Qyriad, 2021-08-04:
I'm not associated with Great Scott Gadgets anymore, and honestly don't even remember most of that code. Consider it your own work; no need to attribute me 🙂

gsigh, 2023-10-29:
sigrok.org mainline now includes GreatFET One support. Took the Qyriad implementation and massaged it to help mainline this approach.

Commits https://sigrok.org/gitaction/libsigrok.git/b4645b0..0dc36c9

Device wiki page https://sigrok.org/wiki/GreatFET_One

Current state:
- Would not work with 2018 firmware. Was created and tested with v2021.2.1 firmware. It's easily available with 'pip install'.
- Supports up to 8 channels at rates up to 40.8MHz, higher rates with fewer channels (up to one channel at 204MHz).
- Does not include support for soft triggers (would like to have this in common code and not repeat non-trivial logic in many drivers).
- Prepares support for 16 channels but firmware support is still under construction.