greatscottgadgets/greatfet #116: Chipcon debug command-line tool

mossmann, 2018-07-10:
Support flash dump, flash erase, flash write, flash verify, peek, and poke.

GravesJake, 2020-08-17:
This was included in #335

Qyriad, 2020-08-20:
#335 included a Python API for Chipcon debugging. This issue is about a command-line tool. i.e.:
```bash
$ greatfet chipcon flash -w file.bin
```

GravesJake, 2020-08-25:
That's right, I misremembered.

Qyriad, 2020-09-14:
Closed with #348.

closed 2020-09-14 by Qyriad

atlas0fd00m, 2020-09-29:
why does this not work with current pip installed version?

```$ greatfet chipcon flash -w file.bin
ERROR: Unsupported subcommand 'chipcon'.
Check to ensure the package providing the subcommand is installed.
```

atlas0fd00m, 2020-09-29:
oh, i see, i thought this was added 2 years ago, didn't see that it was just merged in 2 weeks ago.
thanks all!
@