greatscottgadgets/greatfet #138: Pin manager in GreatFET firmware

dominicgs, 2018-08-02:
It would be nice to have a method of managing and keeping track of pin assignments / settings / functions.  I think this could be implemented as a pin manager that will allow us to set / query pin settings.

mossmann, 2018-08-03:
The source of truth for pin assignments is the SCU registers, so I think the pin manager core should be in firmware.

mnaberez, 2018-09-20:
A pin manager running in the firmware should remember how the host has configured it.  The host should be able to disconnect and reconnect without disturbing any configuration.  For example, if you are running a Python script that uses GPIO pins, it should be possible to exit this script and then open a test panel GUI to observe those pins without disturbing them.  If your Python script is using SPI, you should be able to exit your script and run a program to observe SPI data without having to give all the SPI settings again.  

The pin manager should prevent mistakes and silent failures.  A pin may have multiple modes in the SCU, or a pin may be used by a various tasks running on the GreatFET firmware.  For example, if the user has configured a pin for UART and a firmware task is receiving UART data, the pin manager should produce a friendly error if the user then tries to use the same pin as GPIO without reconfiguring it.  Similarly, there is an open issue to add MicroPython support.  If MicroPython is running and using a pin, then trying to use that same pin from the host should produce a friendly error.  

USB APIs should interact with the pin manager rather than the peripheral registers directly.  If the pin manager is consulted whenever the host needs a pin, it can track all state and can handle starting/stopping tasks related to a pin.  If the LPC registers are exposed directly, it becomes difficult to do the sorts of higher-level things described here.

closed 2018-12-19 by ktemkin