greatscottgadgets/greatfet #141: Add support for a self-test mode.

ktemkin, 2018-08-18:
Provide the capability for a USB-orchestrated self-test mode, which should include:

* Debug logging, a la `dmesg`
* A self-test option that reports overall device/peripheral health targeted towards end-users -- as much as is possible.

mossmann, 2019-02-15:
self-test ideas from notes of prior conversation:

- millisecond timer from USB can be used to test 12 MHz oscillator
- test X1 against IRC
- low-speed emergency mode if X1 doesn't work
- capacitive coupling to ADC pins
- GPIO/ADC pins self-test
- flip all GPIO on and off and make sure still alive
- measure band gap with ADC to measure VCC
- test LEDs
- test DFU button (report state changes to console)
- GPIO self-test of pull-up/pull-down
  - use GPIO internal resistors for test
- detect USB1_VBUS
- test USB1 in device mode
- test USB1 load switch
- test SPIFI
  - detect SPIFI ID

ktemkin, 2019-09-17:
updated the last comment to have checkboxes so we can track individual ideals