greatscottgadgets/greatfet #148: ImportError: cannot import name GreatFET

xairy, 2018-09-22:
Hi!

I've built and installed GreatFET host tools while following this tutorial:
https://github.com/greatscottgadgets/greatfet/tree/master/firmware#building-and-installing-to-flash

However when I try to run any of them I get:

```
$ greatfet_info 
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_info", line 9, in <module>
    load_entry_point('GreatFET==0.0', 'console_scripts', 'greatfet_info')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/GreatFET-0.0-py2.7.egg/greatfet/commands/greatfet_info.py", line 10, in <module>
    from greatfet import GreatFET
ImportError: cannot import name GreatFET
```

At the same time I can import this module without any issues from an interactive python session:

```
$ python
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from greatfet import GreatFET
>>> type(GreatFET)
<type 'function'>
```

What could be the issue and how do I fix this?

dominicgs, 2018-09-27:
I've played around with some installation paths and python versions and I'm unable to recreate this issue.  Could you try a couple of things to help us debug it?

1) I'd like to know the last line of the output is when you do this:
```
python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import greatfet
>>> print greatget.__file__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'greatget' is not defined
>>> print greatfet.__file__
/usr/local/lib/python2.7/dist-packages/GreatFET-0.0-py2.7.egg/greatfet/__init__.pyc
```

2) Can you try running it with an explicit version of python? `python2.7 /usr/local/bin/greatfet_info`

3) If you have python 3 installed, could you try building and installing the GreatFET host code using python 3?

xairy, 2018-09-27:
1. Sure:

```
$ python
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import greatfet
>>> print greatfet.__file__
/usr/local/lib/python2.7/dist-packages/GreatFET-0.0-py2.7.egg/greatfet/__init__.pyc
```

2. Same error.

3. python3 works perfectly.

dominicgs, 2018-09-28:
Thank you.  I'll look at this some more in the next week or so, but for now I think python 3 is a viable work around.

ktemkin, 2018-11-11:
All affected: can you pull from master, re-install the host tools, and try again? This should be fixed-- I'll re-open if anyone reports still being affected.

closed 2018-11-11 by ktemkin