greatscottgadgets/greatfet #159: Flashing greatfet_usb via DFU mode (with flash_usb stub)

ikarus23, 2018-10-05:
Hi,
I want to flash my greatfet back to its "normal" firmware `greatfet_usb`. In order to do so, I need the `-d` and `--dfu-stub` option of the `greatfet_firmware` command, I presume. If I'm not mistaken, `flash_stub` should be a DFU stub for flashing the `greatfet_usb` via DFU. However, if I try to build the `flash_stub` the `cmake` commands fails.
```
cd firmware/flash_stub
mkdir build
cd build
cmake .. -DBOARD=GREATFET_ONE

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
-- Configuring done
CMake Error at /home/user/greatfet/firmware/cmake/greatfet-common.cmake:147 (add_library):
  Cannot find source file:

    ../greatfet_usb/usb_api_board.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:39 (DeclareTargets)

CMake Error at /home/user/greatfet/firmware/cmake/greatfet-common.cmake:172 (add_executable):
  No SOURCES given to target: flash_stub_dfu.elf
Call Stack (most recent call first):
  CMakeLists.txt:39 (DeclareTargets)

[...]
```
Am I doing something wrong?

dominicgs, 2018-10-06:
You aren't doing anything wrong, the flash stub is temporarily broken.  We've been giving the USB API an overhaul and the flash stub hasn't been moved over yet.  More information here: https://pairlist3.pair.net/pipermail/greatfet/2018-September/000013.html

If you build `greatfet_usb` and run `make greatfet_usb-program`, you'll DFU the main firmware on to your GreatFET.  You can then use 'make greatfet_usb-flash" to write it to the flash chip.

ktemkin, 2018-10-10:
Yep. The flash stub should be fixed relatively soon; and eventually, I suspect it'll be the preferred way to interact with a device that's in DFU mode.

My near-term plans include a rewrite of the flash stub to be a single generic stub that works with a variety of related boards -- including the HackRF, the Xplorer, the rad1obadge, and etc. Ideally, the stub will automatically detect which board it's running on and present identification information to the host automatically. :)