greatscottgadgets/greatfet #164: Python errors

Renaud-Pagin, 2018-11-09:
I try since this moorning to make it working with python2.7 or python 3.0

Linux distro : Parrot Debian based
Hardware   : NXP LPC43xx

I compiled flash and can flash the board in DFU mode, but once flashed, i can not use host tools to reflash / request data informations.

I followed the step by step 2 times ... and retry but i do not see any error during the build / install process.

I continue to find ... if some one has one idea :-)

Regards,
Renaud

On python2.7 

`$sudo python2.7 /usr/local/bin/greatfet_info 
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_info", line 11, in <module>
    load_entry_point('GreatFET==0.0', 'console_scripts', 'greatfet_info')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 484, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2714, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2332, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2338, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/GreatFET-0.0-py2.7.egg/greatfet/__init__.py", line 4, in <module>
    from .greatfet import GreatFET
  File "/usr/local/lib/python2.7/dist-packages/GreatFET-0.0-py2.7.egg/greatfet/greatfet.py", line 5, in <module>
    from .board import GreatFETBoard
  File "/usr/local/lib/python2.7/dist-packages/GreatFET-0.0-py2.7.egg/greatfet/board.py", line 18, in <module>
    from pygreat.board import GreatBoard
  File "build/bdist.linux-x86_64/egg/pygreat/board.py", line 16, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pygreat-0.0-py2.7.egg/pygreat/comms.py", line 508
    timeout=1000, encoding=None, max_response_length=4096, 
`

With Python3

`$sudo python3 /usr/local/bin/greatfet_info 
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_info", line 11, in <module>
    load_entry_point('GreatFET==0.0', 'console_scripts', 'greatfet_info')()
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/commands/greatfet_info.py", line 87, in main
    devices = GreatFET(find_all=True)
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/greatfet.py", line 29, in GreatFET
    return GreatFETBoard.autodetect_all(**board_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat-0.0-py3.6.egg/pygreat/board.py", line 96, in autodetect_all
  File "/usr/local/lib/python3.6/dist-packages/pygreat-0.0-py3.6.egg/pygreat/board.py", line 142, in all_accepted_devices
  File "/usr/local/lib/python3.6/dist-packages/pygreat-0.0-py3.6.egg/pygreat/board.py", line 159, in accepts_connected_device
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/boards/xplorer.py", line 25, in __init__
    self.onboard_flash = SPIFlash(self, device_id=0x15, pages=16384, maximum_address=0x3FFFFF)
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/peripherals/spi_flash.py", line 29, in __init__
    self.api = board.apis.spi_flash
AttributeError: 'CommsApiCollection' object has no attribute 'spi_flash'
`

ktemkin, 2018-11-09:
For now, `master` is a bit unstable as we work to move everything over to the new communications protocol. The NXP Xplorer board definition still needs to be updated in the host API. :)

I'll create an issue and hopefully get to poking that relatively soon. In the interim, you're welcome to use the commit just before we moved to the new comms protocol, which you can find in the `pre-libgreat` tag.

Renaud-Pagin, 2018-11-10:
Here is the result of my testing with tag "pre-libgreat"

It still have some issue with python 2.7 ( can not import GreatFET ... , circular dependancies ?? ).

But i confirm that under python 3.6 i successfully flashed my NXP Xplorer, first in DFU mode then in Flash mode.

> 
(py-env) ┌─[✗]─[linux@parrot]─[~/Works/py-env/greatfet/firmware/greatfet_usb/build]
└──╼ $sudo make greatfet_usb-flash
[  3%] Built target greatfet_usb_m0.elf
[  3%] Built target greatfet_usb_m0.bin
[ 98%] Built target OBJ_FILES_greatfet_usb
[100%] Built target greatfet_usb.elf
[100%] Built target greatfet_usb.bin
Trying to find a GreatFET device...
NXP Xplorer found. (Serial number: 0000000000000000ffffffffffffffff)
Writing data to SPI flash...
Written 38156 bytes of 38156.
Write complete!
Resetting GreatFET...
Reset complete!
[100%] Built target greatfet_usb-flash
(py-env) ┌─[linux@parrot]─[~/Works/py-env/greatfet/firmware/greatfet_usb/build]
> 

> (py-env) ┌─[✗]─[linux@parrot]─[~/Works/py-env/greatfet/firmware/greatfet_usb/build]
└──╼ $sudo greatfet_info 
Found a NXP Xplorer!
  Board ID: 1
  Firmware version: git-a76e0cc
  Part ID: 300a00a0444f53
  Serial number: 0000000000000000ffffffffffffffff
 >

ttepatti, 2020-02-28:
Hello, I'm still running into this issue in 2020 :)

I'm using Python 3.6.9 on Ubuntu 18.04.4 LTS

I have basically the exact same issue as Renaud. I can flash the NXP Xplorer initially in DFU mode, but afterwards, can no longer issue any commands to the board. Running "gf info" results in a "KeyError: 'spi_flash'" error with the board plugged in.