greatscottgadgets/greatfet #167: TravisCI: install and run the host tools

ktemkin, 2018-11-11:
Ideally, we'd have the TravisCI script run the install for the host tools, and maybe check to ensure that we can run one of the scripts-- e.g. that `greatfet info` runs enough to not find a GreatFET.

This would help to prevent issues like #148 from being introduced. :)

mnaberez, 2018-11-15:
Consider running a tool like [pyflakes](https://pypi.org/project/pyflakes/) as part of the host tools CI to check code (not style).  It will help find common mistakes like [missing imports](https://github.com/greatscottgadgets/greatfet/commit/5d515f1857e025fe6359253cef6a47b1cbbc490c#diff-0cddee9ce0aafd6c9e722ebc35825ab9).

ktemkin, 2019-08-22:
The current travis script does exercise installation of the host tools. Closing as done.

closed 2019-08-22 by ktemkin