greatscottgadgets/greatfet #178: AttributeError: 'GreatFETOne' object has no attribute 'vendor_request_out'

gorzilla, 2018-12-08:
I get an error trying to run i2c scan. I'm running the latest firmware and host tools.

```
$ greatfet info
Found a GreatFET One!
  Board ID: 0
  Firmware version: git-67a464f
  Part ID: 300a00a04e4f6100
  Serial number: 000074865d930634d8b
```

```
$ greatfet_i2c -z
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_i2c", line 11, in <module>
    load_entry_point('GreatFET==0.0', 'console_scripts', 'greatfet_i2c')()
  File "/usr/local/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/commands/greatfet_i2c.py", line 44, in main
    scan(device, log_function)
  File "/usr/local/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/commands/greatfet_i2c.py", line 50, in scan
    i2c_bus = I2CBus(device)
  File "/usr/local/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/peripherals/i2c_bus.py", line 42, in __init__
    board.vendor_request_out(vendor_requests.I2C_START, value=duty_cycle_count)
AttributeError: 'GreatFETOne' object has no attribute 'vendor_request_out'
```

GravesJake, 2018-12-10:
See 074a425

closed 2018-12-10 by GravesJake

GravesJake, 2018-12-11:
Should be fixed, try pulling and installing the latest host tools.