greatscottgadgets/greatfet #195: USb detach/attach on warm reset - arbitrary delay needed

colinoflynn, 2018-12-27:
When the device does a warm reset (after calling a reset or a FW update), it will do a detach/reattach in the `init_usb0()` call within `greatfet_usb.c`.

On some systems a relatively large delay is needed between detach and reattach. I had to add `delay_us(500000)` before the call to `usb_run()` for it to be detected on my system after a warm reset - which is using Windows 7 + a VirtualBox VM. That is probably (hopefully?) towards the upper end of such needs due to both Windows and the VM.

This can be made less hacky by detecting the warm reset (looking for device still attached on entry) and only doing the delay then.

I'm not sure if there is a better solution however, so wanted to post an issue for those with more USB experience before I submit a PR.

ktemkin, 2019-02-15:
Sorry about the delay on getting back to you on this-- I saw this earlier, but then it wound up buried behind a bunch of other USB-related tasks. :)

At a glance, this issue doesn't seem to be on our end -- does it reproduce reliably on VirtualBox? If so, I'll set up a VM and take a look at the actual USB traffic and see if there are any clues on our end at some point in the future.

I don't think adding an arbitrarily large delay is necessarily the right solution; but I do want to try and get this working reliably in setups like yours. If we need to come up with workarounds, I'll try and make sure we have some functionality that allows this to work with VirtualBox.

Let me know on the reproducibility and I'll figure out next steps.

colinoflynn, 2019-02-15:
No issue at all on delay - I hadn't looked at this since then ;-) Are you using a Windows host? From what I recall on other USB stuff Windows sometimes did have such delay requirements, but this is going from memory some time ago.

ktemkin, 2019-09-17:
I haven't had this issue rapidly attaching/detaching GreatFETs (incl during reset) on VirtualBox in a while; so I'm going to close this. If you run into it again, feel free to re-open. :)

closed 2019-09-17 by ktemkin