greatscottgadgets/greatfet #200: Python errors when cli tool cannot communicate

kg6hum, 2019-01-09:
Hardware: GreatFET One
Firmware: git-62d6747
Host Software: git-62d6747
Python Version: 3.6.7

When I do not have a GreatFET plugged in, or if the gf cli tool has trouble communicating to the GreatFET while it is connected, I get a python error for an undefined Exception and an issue importing the Exception.

I can reproduce the issue with the following commands:
gf i2c -z
gf shell

It does not happen with "gf info"

$ gf i2c -z
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/commands/greatfet_i2c.py", line 33, in main
    device = GreatFET(serial_number=args.serial)
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/greatfet.py", line 31, in GreatFET
    return GreatFETBoard.autodetect(**board_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat-0.0-py3.6.egg/pygreat/board.py", line 66, in autodetect
  File "/usr/local/lib/python3.6/dist-packages/pygreat-0.0-py3.6.egg/pygreat/board.py", line 162, in accepts_connected_device
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/boards/legacy.py", line 63, in __init__
    raise DeviceNotFoundError()
NameError: name 'DeviceNotFoundError' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/greatfet_i2c", line 11, in <module>
    load_entry_point('GreatFET==0.0', 'console_scripts', 'greatfet_i2c')()
  File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/commands/greatfet_i2c.py", line 35, in main
    except greatfet.errors.DeviceNotFoundError:
AttributeError: module 'greatfet' has no attribute 'errors'

mnaberez, 2019-01-09:
> Host Software: git-62d6747
...
> File "/usr/local/lib/python3.6/dist-packages/GreatFET-0.0-py3.6.egg/greatfet/boards/legacy.py", line 63, in init raise DeviceNotFoundError() NameError: name 'DeviceNotFoundError' is not defined

Here it is on that rev (which is the current master at this moment):

```
greatfet$ git checkout 62d6747 
greatfet$ pyflakes host/greatfet/boards/legacy.py 
host/greatfet/boards/legacy.py:56: undefined name 'LIBUSB_PIPE_ERROR'
host/greatfet/boards/legacy.py:57: undefined name 'DeviceNotFoundError'
host/greatfet/boards/legacy.py:63: undefined name 'DeviceNotFoundError'
host/greatfet/boards/legacy.py:70: undefined name 'DeviceNotFoundError'
host/greatfet/boards/legacy.py:74: undefined name 'DeviceNotFoundError'
host/greatfet/boards/legacy.py:167: local variable 'e' is assigned to but never used
host/greatfet/boards/legacy.py:171: local variable 'connected' is assigned to but never used
host/greatfet/boards/legacy.py:173: undefined name 'time'
host/greatfet/boards/legacy.py:173: undefined name 'RECONNECT_DELAY'
host/greatfet/boards/legacy.py:266: 'usb' imported but unused
host/greatfet/boards/legacy.py:268: undefined name 'vendor_requests'
```

Fixing those missing imports should fix the error.  

There's an open issue (#167) to check the host tools during CI.  I suggest running `pyflakes` during CI.  It would have caught this one.

kg6hum, 2019-01-10:
> Which version of firmware are you using? It looks like you're hitting the legacy board version in the host python, so I think it might be worth using DFU mode to update your firmware to the latest, if you aren't already running it?

As mnaberez noted above, this is with matching firmware and software versions from current master, commit 62d6747.  It is easily reproducible for me by just running "gf shell" or "gf i2c -z" without a GreatFET connected.  I would guess that it is checking for a legacy device since it is not finding a current one, and it is reaching some code branches with undefined names.

ktemkin, 2019-03-07:
This shouldn't be an issue, any longer -- should have been fixed with some of the libgreat changes. Closing for now -- I'll re-open if anyone is experiencing issues still on master.

closed 2019-03-07 by ktemkin