greatscottgadgets/greatfet #205: automate release archive production and versioning

mossmann, 2019-01-17:
The manual release process is here:
https://github.com/greatscottgadgets/greatfet/wiki/Release-Process

We want matching version strings to be added to setup.py and to firmware, generated from git tags and/or a top level version file.

dominicgs, 2019-01-19:
An option is to use `git describe` which gives you a string based on the latest tag.  If you've checked out the tag you get the tag name, otherwise you get something like `v2018.01.1-239-ge5f117d`

ktemkin, 2019-03-08:
An initial stab at this is located in the `release_prep` branch of my fork.

See especially:
https://github.com/greatscottgadgets/greatfet/compare/master...ktemkin:release_prep?expand=1#diff-b67911656ef5d18c4ae36cb6741b7965R77

Steps for release with this are:

- install `git-archive-all`, possibly with `pip3 install git-archive-all`
- copy updated `RELEASENOTES` into the tree
- run `make prepare_release VERSION=YYYY.MM.NN`
- inspect the output, if you like it, follow the commands the makefile recommended (push the tag, push the pypi package) and upload the archives that are now in `release-files`.

ktemkin, 2019-05-04:
Update: I submitted an upstream PR to a project we use in our *non-release* versioning code; hopefully that fixes a couple of things.

Linking it here so we can track it:

https://github.com/vivin/better-setuptools-git-version/pull/1

closed 2019-06-07 by ktemkin