greatscottgadgets/greatfet #212: Usage/user-facing documentation

ktemkin, 2019-02-15:
There’s currently a dearth of user-facing documentation and guides for GreatFET. We should rectify this; and this is especially important as more hardware winds up in end-user hands.

The comments for this issue are probably a good place to ideate about things we might want to document / guides we might want to create.

mossmann, 2019-02-15:
start greatfet-tutorials repo and use gh-pages branch

Getting Started with GreatFET:
- plug in GreatFET One and try command-line tools (start with gf info)
- blink LED from command-line
- blink LED from Python
- next steps  (pointers to other guides)

Logic Analysis:
- gf logic example, write to VCD file and load in pulseview
- (future) real-time sigrok example

SPI:
- working with a SPI flash chip

I2C:
- scan
- I/O expander example

ktemkin, 2019-09-17:
converted the bullet point list here to use fancy github checkboxes