greatscottgadgets/greatfet #225: adding jtag  pinout searching functions similar to jtagulator or jtagenum?

nihilisticeevee, 2019-04-02:
Feature request to add functions to search for jtag pins for unknown pinouts. If this feature was added, it would be a must have tool.

ktemkin, 2019-04-03:
This is definitely something we'd like to have -- and would be an even more useful feature when combined with some of the neighbor boards that are in the works.

Thanks for the suggestion -- marking this as a 'probable' enhancement for the future. (This would probably also make a nice jumping-in point for anyone who wants to get their feet wet re: GreatFET firmware development. :))

jrozner, 2019-04-08:
I spoke to Dominic about this last year and expressed an interest in working on it. I don’t really have any free time until after May but if nobody takes this before then I plan to. Was going to look at designing a neighbor to go with it to provide all the input protection that the jtagulator has.

ktemkin, 2019-04-08:
@jrozner Neat! If the timing works out, that sounds great.

     Was going to look at designing a neighbor to go with it to provide all the input protection that the jtagulator has.

There's a neighbor in progress (Foxglove) that will provide some digital front-end support (it's intended to be a general digital frontend for e.g. logic analyzer use); this might provide all the input protection we'd need.