greatscottgadgets/greatfet #231: gcc/arm-none-eabi make failure

yhetti, 2019-05-03:
I had to work around a different bug in gcc/arm-none-eabi for a different piece of hardware. The solution to which was to install the gcc/arm-none-eabi PPA [ppa:team-gcc-arm-embedded/ppa].
After which I went back and tried to 'make' the greatfet firmware and got the following errors:

...
...
GENHDR  include/libopencm3/efm32/efm32g/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32tg/irq.yaml
  BUILD   lib/stm32/f0
make[4]: *** No rule to make target '/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h', needed by 'flash.o'.  Stop.
Makefile:67: recipe for target 'lib/stm32/f0' failed
make[3]: *** [lib/stm32/f0] Error 2
greatfet_usb/CMakeFiles/libopencm3.dir/build.make:110: recipe for target 'greatfet_usb/libopencm3-prefix/src/libopencm3-stamp/libopencm3-build' failed
make[2]: *** [greatfet_usb/libopencm3-prefix/src/libopencm3-stamp/libopencm3-build] Error 2
CMakeFiles/Makefile2:291: recipe for target 'greatfet_usb/CMakeFiles/libopencm3.dir/all' failed
make[1]: *** [greatfet_usb/CMakeFiles/libopencm3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Qyriad, 2019-05-06:
Looks like `make clean`ing in `firmware/libopencm3` fixes this.
I imagine it's probably due to libopencm3 being configured with the arm-none-eabi-gcc package, which is gcc 6.3.1, while gcc-arm-embedded is gcc 7.3.1.

ktemkin, 2019-05-07:
Yeah, this is because generated Makefiles in *libopencm3* wind up containing full paths to the standard libraries. 

I'd normally say that we should add a `make clean` for *libopencm3* to the `make clean` step in our CMake, but we really should just get rid of the last few remnants of *libopencm3* and move entirely to `libgreat`.

ktemkin, 2019-05-07:
I can confirm that running `make clean` from *libopencm3* in @yhetti's build directory fixed this, as I poked his computer a bit and the build is working now. :)

I'm closing this, as it's not really an issue with GreatFET or the arm-embedded toolchain; and it's not worth accounting for in libopencm3 currently. Feel free to re-open if things break further.

closed 2019-05-07 by ktemkin