greatscottgadgets/greatfet #236: Inconsistent "Operation timed out" with Greatfet

szczys, 2019-05-09:
After following the udev rules install outlined in [Getting Started](https://greatscottgadgets.github.io/greatfet-tutorials/getting-started.html) I am sometimes able to connect to Greatfet, or connection is inconsistent.

I can get the info, but when I try shell I get Operation Timed Out. I thought this was a udev problem and started writing this issue that way, but it seems like if I keep trying the gf shell command, eventually I get one.

```
 $ greatfet info
Found a GreatFET One!
  Board ID: 0
  Firmware version: 2018.12.1
  Part ID: a0000a305d4f6c
  Serial number: 000057cc67e6319b5257
 
 $ greatfet shell
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_shell", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/greatfet/commands/greatfet_shell.py", line 27, in main
    gf = parser.find_specified_device()
  File "/usr/local/lib/python3.6/dist-packages/greatfet/utils.py", line 111, in find_specified_device
    device = self._find_greatfet(args)
  File "/usr/local/lib/python3.6/dist-packages/greatfet/utils.py", line 183, in _find_greatfet
    return GreatFET(serial_number=args.serial)
  File "/usr/local/lib/python3.6/dist-packages/greatfet/greatfet.py", line 31, in GreatFET
    return GreatFETBoard.autodetect(**board_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 77, in autodetect
    board.initialize_apis()
  File "/usr/local/lib/python3.6/dist-packages/greatfet/boards/one.py", line 135, in initialize_apis
    super(GreatFETOne, self).initialize_apis()
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 264, in initialize_apis
    self.comms.run_autoenumeration()
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 124, in run_autoenumeration
    self._generate_object_for_class(class_number)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 159, in _generate_object_for_class
    attrs = self._generate_rpc_verbs_for_class(class_number, class_name)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 211, in _generate_rpc_verbs_for_class
    name            = core_api.get_verb_name(class_number, verb_number)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 1088, in method
    timeout=timeout, max_response_length=max_response_length, encoding=encoding, *arguments)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 1243, in execute_command
    out_format, *arguments, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 761, in execute_command
    None, max_response_length, comms_timeout, pretty_name, rephrase_errors)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms_backends/usb.py", line 281, in execute_raw_command
    self.LIBGREAT_REQUEST_NUMBER, self.LIBGREAT_VALUE_EXECUTE, flags, max_response_length, comms_timeout)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
```
Looking into this, here's my dmesg:
```
[378064.487574] usb 1-10: USB disconnect, device number 8
[378066.352077] usb 1-10: new high-speed USB device number 9 using xhci_hcd
[378066.521947] usb 1-10: New USB device found, idVendor=1d50, idProduct=60e6
[378066.521951] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[378066.521953] usb 1-10: Product: GreatFET
[378066.521955] usb 1-10: Manufacturer: Great Scott Gadgets
[378066.521957] usb 1-10: SerialNumber: 000000000000000057cc67e6319b5257
```
I'm in the plugdev group:
```
 $ cat /etc/group | grep plugdev
plugdev:x:46:mike
```
OpenMoko shows up in lsusb (the NXP DFU does not but I don't think it should unless you're doing a fw update)
```

Running through the command I get about 50% instances where operation times out.

System:
```
Python3.6
Linux 4.15.0-48-generic #51-Ubuntu SMP Wed Apr 3 08:28:49 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

mossmann, 2019-05-10:
Is this happening after the firmware update you did in #237?

szczys, 2019-05-10:
Yes.

mossmann, 2019-05-10:
Is the heartbeat on LED1 working consistently even when you get these errors?

szczys, 2019-05-10:
I'm going to call this one a cable problem. I've been using a USB extension cable and just tested it connecting the cable directly to the port. This has solved my timeout problems. Sorry I didn't think to test that earlier.

closed 2019-05-10 by szczys