greatscottgadgets/greatfet #239: Building greatfet_usb

coley7, 2019-05-15:
How to handle this issue when building greatfet_usb

[  8%] Built target libopencm3
[ 14%] Built target libgreat_module_bsp
[ 23%] Built target libgreat
[ 24%] Built target libgreat_module_debug-backtrace
[ 28%] Built target libgreat_module_comms
[ 28%] Built target libgreat_module_ad970x
[ 30%] Built target libgreat_module_allocator
[ 35%] Built target libgreat_module_usb
[ 36%] Built target libgreat_module_gpio
[ 37%] Built target libgreat_module_usb_comms
[ 79%] Built target greatfet_usb
[ 97%] Built target libgreatfet
[ 98%] Linking C executable greatfet_usb.dfu.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(scu.o) uses VFP register arguments, greatfet_usb.dfu.elf does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(scu.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(i2c.o) uses VFP register arguments, greatfet_usb.dfu.elf does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(i2c.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(timer.o) uses VFP register arguments, greatfet_usb.dfu.elf does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(timer.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(vector.o) uses VFP register arguments, greatfet_usb.dfu.elf does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(vector.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(nvic.o) uses VFP register arguments, greatfet_usb.dfu.elf does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(nvic.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(sync.o) uses VFP register arguments, greatfet_usb.dfu.elf does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tools/greatfet/firmware/greatfet_usb/../../firmware/libopencm3/lib/libopencm3_lpc43xx.a(sync.o)
collect2: error: ld returned 1 exit status
CMakeFiles/greatfet_usb.dfu.elf.dir/build.make:226: recipe for target 'greatfet_usb.dfu.elf' failed
make[2]: *** [greatfet_usb.dfu.elf] Error 1
CMakeFiles/Makefile2:219: recipe for target 'CMakeFiles/greatfet_usb.dfu.elf.dir/all' failed
make[1]: *** [CMakeFiles/greatfet_usb.dfu.elf.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

ktemkin, 2019-05-15:
That looks like your `libopemcm3` and your main GreatFET firmware were built using different compiler configurations. Two environment questions:
1. Have you upgraded your compiler recently, or switched compilers from the GNU-gcc to the ARM-gcc toolchain? 
2. Are you using Ubuntu, or a Ubuntu-like distribution? There have been various issues with the way they provide libraries with their arm-none compilers.

Either way, I'd try running `make clean` in the `firmware/libopencm3` folder, then `make clean` in your firmware build directory, and see if the problem resolves itself.

coley7, 2019-05-17:
I tried it on a Ubuntu 18.04 with current releases. There it didn't work
It worked only on a non current Debian version... 

Is there a way to provide this more stable?

Qyriad, 2019-05-21:
It built fine for me in containers for the latest Debian and Ubuntu releases (9 Stretch, and 18.10 Cosmic Cuttlefish, respectively), but I was able to reproduce this in an Ubuntu 18.04 LTS (Bionic Beaver) container, with the `gcc-arm-none-eabi` package.

Changing to `gcc-arm-embedded` (`ppa:team-gcc-arm-embedded/ppa`) fixed it, for me.
Can you try switching to that package and see if it works? Remember to `make clean` in `firmware/libopencm3` after changing toolchains.

coley7, 2019-05-21:
well done! It's working!
Thank you very much.

closed 2019-05-21 by coley7