greatscottgadgets/greatfet #245: Finalize udev rule installation method

ktemkin, 2019-06-07:
We should make it easy for users to install the udev rules, e.g. if they install from pip.

yuvadm, 2019-07-25:
Installing udev rules is usually handled by the system package manager (e.g. [how it's done in Arch](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=greatfet-git))

Maybe the right way to handle this is to print a nice hint in case a board is not found due to insufficient permissions on the device?

ktemkin, 2019-07-25:
For distribution provided packages, the maintainers including the udev rules in each package is absolutely the way to go.

For users whose distribution doesn't provide a package (or an equivalent to the AUR), installing from `pip` is a good option (especially with `--user`). It'd be nice to be able to support them by e.g. having a greatfet subcommand that ensures e.g. udev rules are installed and working, or the proper libusb userland driver is installed on Windows.

I'd like a script that does some really basic distro detection and then writes udev rules according to local conventions (e.g. `plugdev` on Debian-ish distros, or tagging with `uaccess` / `udev-acl` for systemd-flavored-udev systems).

commit 746602c2 (2019-08-09): host: add gf host-tools command with --ensure-access

`gf host-tools --ensure-access` will attempt to install udev rules on
Linux based on whether the current system has systemd: uaccess rules if
it does, plugdev if it doesn't.

Closes #245.
closed 2019-08-09 by ktemkin