greatscottgadgets/greatfet #247: fix greatfet_adc.py

mossmann, 2019-06-07:
It currently appears to be trying but failing to stream.  It should do a one-shot reading instead of streaming by default.  Streaming should work.  It shouldn't dump so much debug output by default. The -a option looks like it should go away.  Check the other options for sanity too.

mossmann, 2019-06-12:
Streaming now appears to be working (in Python2 on Ubuntu 19.04).  Perhaps I misinterpreted the results when I opened this issue, or perhaps it only works on some platforms.

mossmann, 2019-06-19:
Streaming also works in Python3 on Ubuntu 19.04.

ktemkin, 2019-07-25:
Yep -- it looks like streaming is working, but the tool's usage is pretty counter-intuitive. This should definitely be rewritten soon.

ktemkin, 2019-07-25:
Started some work on this. The tool now, by default, outputs:

```
# gf adc                                 
3.18076171875V
```

(I'd love to truncate the amount of significant digits we have on there to better match the ADC accuracy, but floating point makes that a bit messy.)

ktemkin, 2019-08-13:
The branch https://github.com/ktemkin/greatfet/tree/adc_rewrite has this pretty much done, but there are a few improvements I'd love to see as this is merged:

- An `ADC` class under "peripherals" in the host code. Ideally, this would have arguments that let you select the ADC parameters (especially which ADC), and maybe a function that lets you adjust the number of significant bits on the fly.
- An instance of the above class as `greatfet.adc` for the default ADC pin (ADC0, channel 0).
- An option for the command-line tool to return a raw sample instead of a voltage.
- An API / option for reading samples continuously into a file or stdout, like the old utility had.

Some things that might be nice:
- An option for the command-line tool to return something easily machine parse-able.
- An option for the command-line tool to grab a fixed number of samples and print them in the same format otherwise specified (e.g. voltage, raw, to file, etc.).

Assigning to @Qyriad for those final improvements. :)

closed 2019-08-22 by ktemkin