greatscottgadgets/greatfet #248: greatfet info command Input/Output Error

straithe, 2019-06-09:
### Steps to reproduce
1.  Followed the tutorial up to the step "Check your GreatFET One".

### Expected behaviour
I expected the command "greatfet info" to show my info about my GreatFet.

### Actual behaviour
I got a traceback that ended in "usb.core.USBError: [Errno 5] Input/Output Error"

### Attempted fixes
* Restarted my computer between attempts.
* Ran the greatfet info command with sudo and without. 

### Version information
**Operating system**:
Ubuntu 18.04.02

**GreatFET Firmware version**:
2018.12.1

### Output
:~$ sudo greatfet info
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_info", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/greatfet/commands/greatfet_info.py", line 94, in main
    devices = GreatFET(find_all=True)
  File "/usr/local/lib/python3.6/dist-packages/greatfet/greatfet.py", line 29, in GreatFET
    return GreatFETBoard.autodetect_all(**board_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 105, in autodetect_all
    subclass_devices = subclass.all_accepted_devices(**device_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 151, in all_accepted_devices
    if cls.accepts_connected_device(**identifiers):
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 179, in accepts_connected_device
    raise e
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 168, in accepts_connected_device
    potential_device = cls(**device_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/greatfet/boards/xplorer.py", line 23, in __init__
    super(NXPXplorer, self).__init__(**device_identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/board.py", line 237, in __init__
    self.comms = CommsBackend.from_device_uri(**self.identifiers)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms.py", line 83, in from_device_uri
    return USBCommsBackend(**device_uri)
  File "/usr/local/lib/python3.6/dist-packages/pygreat/comms_backends/usb.py", line 102, in __init__
    self.device.set_configuration()
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 147, in managed_set_configuration
    self.managed_open()
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error

straithe, 2019-06-09:
In an attempt to solve this issue:
* I did an apt-get install dfu-util. 
* I ran greatfet fw -d --auto.
* The command appeared to work and gave me the output: 
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        52736 bytes
Download done.
dfu-util: unable to read DFU status after completion
Trying to find a GreatFET device...
libgreat compatible board in flash-stub mode found. (Serial number: 000057cc67e630285657)
Writing data to SPI flash...
Written 84812 bytes of 84812.
Write complete!
Resetting GreatFET...
Reset complete!

straithe, 2019-06-09:
I installed from source commit f82c9 and it appeared to work, but when I ran greatfet info again I got the same error message as is in my initial issue comment.

mossmann, 2019-06-12:
I was unable to reproduce this error on a fresh Ubuntu 18.04.02 installation with 2018.12.1 firmware.

evanrichter, 2019-06-13:
This seems like it could be the same error I am getting: https://pairlist3.pair.net/pipermail/greatfet/2019-June/000034.html

Try calling `gf info; gf info`

I bet the second call will work.

Kate Tempkin replied to me suggesting it could be the host computer putting the USB device to sleep as a power saving strategy. I haven't tried that yet though I think her idea is right.

straithe, 2019-06-13:
Just tried it out, and it worked! Thank you, @evanrichter and @ktemkin for the (partial) fix.

Note: As soon as I stop using the GreatFET for a few moments I get the same error, but I can run the gf info command multiple times in a row again and things seem to work out correctly.

evanrichter, 2019-06-13:
Great! glad to get it up and running.
To report back, @ktemkin was completely correct in her assessment. I adjusted my TLP config (`/etc/default/tlp` on arch) to include the line:
`USB_BLACKLIST="1d50:60e6"`
which blacklists the greatfet from usb autosuspend

It looks like #255 could be another workaround for the issue

szczys, 2019-07-02:
I can confirm the same problem on a Dell XPS 13 laptop running Ubuntu 18.04 bionic

Adding the blacklist line mention above to /etc/default/tlp and then running `sudo tlp start` then replugging the GreatFET solved my issue.

vergil-SI, 2019-07-18:
Same here on a Dell XPS 15 running 18.04. The fix @evanrichter mentioned worked for me as well.

ktemkin, 2019-08-22:
> It looks like #255 could be another workaround for the issue.

This is less a workaround and more fixing compatibility with USB suspend on Linux; but it should fix the problem. This should be fixed on master, so I'm going to close the issue.

If anyone here is interested in trying the fix out, you can grab the current build of the GreatFET firmware from: 

https://nightlies.ktemkin.com/greatfet/2019/08/greatfet-2019-08-16-build_463-git_5fd862d.zip

closed 2019-08-22 by ktemkin