greatscottgadgets/greatfet #249: pushd and popd not found

straithe, 2019-06-09:
My bash has popd and pushd, but my /bin/sh does not. I am on Ubuntu 18.04.02.

sudo make install
git submodule update --init
Submodule 'firmware/libopencm3' (https://github.com/dominicgs/libopencm3) registered for path 'firmware/libopencm3'
Submodule 'libgreat' (https://github.com/greatscottgadgets/libgreat) registered for path 'libgreat'
Cloning into '/home/straithe/Desktop/greatfet/firmware/libopencm3'...
Cloning into '/home/straithe/Desktop/greatfet/libgreat'...
Submodule path 'firmware/libopencm3': checked out '1c45abedbce49fe179030287325baf91f7e0d5e1'
Submodule path 'libgreat': checked out 'e14af8d6ef89f28c2fc94c277e23a58ffa55f6dc'
pushd libgreat/host; python setup.py install; popd
/bin/sh: 1: pushd: not found
python: can't open file 'setup.py': [Errno 2] No such file or directory
/bin/sh: 1: popd: not found
Makefile:64: recipe for target 'install' failed
make: *** [install] Error 127

mossmann, 2019-06-12:
I reproduced this on Ubuntu 19.04. The pushd and popd commands are available in bash but not dash (which is what /bin/sh is a  symlink to).

ktemkin, 2019-07-25:
Done, and merged. Thanks for the report. :)

closed 2019-07-25 by ktemkin