greatscottgadgets/greatfet #250: Unknown error while querying info from VMWare Virtual Machine

mteears, 2019-06-11:
I am receiving a series of errors when trying to initiate a connection with the GreatFET in a virtual machine running Kali Linux 19.02.  The error is as follows:

root@kali:~# gf info
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_info", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/greatfet/commands/greatfet_info.py", line 94, in main
    devices = GreatFET(find_all=True)
  File "/usr/local/lib/python3.7/dist-packages/greatfet/greatfet.py", line 29, in GreatFET
    return GreatFETBoard.autodetect_all(**board_identifiers)
  File "/usr/local/lib/python3.7/dist-packages/pygreat/board.py", line 105, in autodetect_all
    subclass_devices = subclass.all_accepted_devices(**device_identifiers)
  File "/usr/local/lib/python3.7/dist-packages/pygreat/board.py", line 151, in all_accepted_devices
    if cls.accepts_connected_device(**identifiers):
  File "/usr/local/lib/python3.7/dist-packages/pygreat/board.py", line 179, in accepts_connected_device
    raise e
  File "/usr/local/lib/python3.7/dist-packages/pygreat/board.py", line 168, in accepts_connected_device
    potential_device = cls(**device_identifiers)
  File "/usr/local/lib/python3.7/dist-packages/greatfet/boards/legacy.py", line 73, in __init__
    self.device.set_configuration()
  File "/usr/local/lib/python3.7/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python3.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/usb/core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 794, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error

The GreatFET is detected with lsusb as follows:
root@kali:~# lsusb
Bus 002 Device 002: ID 1d50:60e6 OpenMoko, Inc. replacement for GoodFET/FaceDancer - GreatFet
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0e0f:0008 VMware, Inc. 
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

mossmann, 2019-06-12:
Can you try `sudo gf info` to check if it is a permissions problem?

mteears, 2019-06-12:
`sudo gf info` returns the same error.

However, when the VM doesn't have the GreatFET connected to it, Device Manager on the Windows Host machine does not have any drivers for the GreatFET device (Code 28).  I believe that this may be causing the communication issues between the Linux VM and the GreatFET.  I will try and reproduce this issue on a machine running linux natively.

mteears, 2019-06-12:
Two different machines running Manjaro natively fail to detect the GreatFET

jsmith79, 2019-06-17:
I am having the same issue when using VM Ware Workstation 12 on a windows 10 host with Ubuntu 18.04 LTS vm, When using on VMware Fusion on MacOS and a Ubuntu 18.04 LTS vm it works ok.

mteears, 2019-06-17:
I installed Ubuntu 19.04 on hardware and no connection issues.

marcusbarbu, 2019-08-13:
I've observed the `No GreatFET found` behavior on both an 18.04 Ubuntu VM in VMWare Player and on a Windows 10 host when running `gf info`

markoHUN, 2021-02-12:
> I've observed the `No GreatFET found` behavior on both an 18.04 Ubuntu VM in VMWare Player and on a Windows 10 host when running `gf info`

in VMWare on the top of the left corner click: player -> removable devices -> connect GreatFet One by manually

meeas, 2022-07-29:
I think it is safe to close this issue.  Every report in this thread was using VMware on Windows 10/11 which meant every case can be explained by #357.

The solution is adding the following line to your VMware virtual machine's VMX config file.

```
usb.quirks.device0 = "0x1d50:0x60e6 allow, skip-reset, skip-refresh, skip-setconfig"
```

I've confirmed this fixes the issue on several different machines.

NOTE: this should not be needed with VMware on Mac or Linux, just VMware on Windows.

martinling, 2022-07-29:
It would be good to understand exactly what "quirk" of GreatFET causes it to require this config line. It may be something we can fix on the firmware side.