greatscottgadgets/greatfet #251: Python2: No module named functools_lru_cache

mossmann, 2019-06-12:
```
$ gf info
Traceback (most recent call last):
  File "/usr/local/bin/gf", line 6, in <module>
    from greatfet.commands.gf import main
  File "/usr/local/lib/python2.7/dist-packages/greatfet/__init__.py", line 3, in <module>
    from .greatfet import GreatFET
  File "/usr/local/lib/python2.7/dist-packages/greatfet/greatfet.py", line 5, in <module>
    from .board import GreatFETBoard
  File "/usr/local/lib/python2.7/dist-packages/greatfet/board.py", line 12, in <module>
    from pygreat.board import GreatBoard
  File "/usr/local/lib/python2.7/dist-packages/pygreat/board.py", line 15, in <module>
    from pygreat.comms import CommsBackend
  File "/usr/local/lib/python2.7/dist-packages/pygreat/comms.py", line 21, in <module>
    from backports.functools_lru_cache import lru_cache as memoize_with_lru_cache
ImportError: No module named functools_lru_cache
```

This happens despite backports.functools-lru-cache having been installed by pip2 as a dependency.  I can find it in /usr/local/lib/python2.7/dist-packages/backports/.

I'm on Ubuntu 19.04.

mossmann, 2019-06-12:
The backports import path does not include /usr/local/lib/python2.7/dist-packages/.

```
$ ipython2
Python 2.7.16 (default, Apr  6 2019, 01:42:57) 
Type "copyright", "credits" or "license" for more information.

IPython 5.8.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import backports

In [2]: backports.__path__
Out[2]: ['/usr/lib/python2.7/dist-packages/backports']

In [3]: import sys

In [4]: sys.path
Out[4]: 
['',
 '/usr/local/bin',
 '/usr/lib/python2.7',
 '/usr/lib/python2.7/plat-x86_64-linux-gnu',
 '/usr/lib/python2.7/lib-tk',
 '/usr/lib/python2.7/lib-old',
 '/usr/lib/python2.7/lib-dynload',
 '/home/mossmann/.local/lib/python2.7/site-packages',
 '/usr/local/lib/python2.7/dist-packages',
 '/usr/lib/python2.7/dist-packages',
 '/usr/local/lib/python2.7/dist-packages/IPython/extensions',
 '/home/mossmann/.ipython']
```

mossmann, 2019-06-12:
workaround:

```
sudo pip2 uninstall backports.functools-lru-cache
sudo apt install python-backports.functools-lru-cache
```

ktemkin, 2019-06-12:
Can you check to see if an apt/dpkg package owns the /use/lib backports, and if so, which one?

It sounds like a backports package was installed with the system package manager; which precludes use of the pip subpackage installed in local. One solution might be to instruct users to install using a pip argument to place packages in a better location (possibly using —user?).

mossmann, 2019-06-12:
configparser is the only other thing in /usr/lib/python2.7/dist-packages/backports

mossmann, 2019-06-12:
`sudo apt remove python-configparser` tells me that it would also remove python-entrypoints and python-keyring.

yuvadm, 2019-07-10:
This error should be fixed by https://github.com/greatscottgadgets/libgreat/pull/5

Qyriad, 2019-08-20:
So this issue is a little bit interesting.

Installing `python-backports.functools-lru-cache` with apt, and then installing greatfet (and libgreat) either with pip or `python setup.py install`, and either with `--user` or not, works just fine.

Installing greatfet and libgreat with `python setup.py install` (`--user` or not), but *without* having installed `python-backports.functools-lru-cache` with apt also works just fine.

But installing with *pip* (`pip install .`), `--user` or not, without installing functools_lru_cache with apt does not work.

I'm thinking just telling users to install `python-backports.functools-lru-cache` with the system package manager might be the way to go until we officially drop Python 2 support.

ktemkin, 2019-08-20:
> I'm thinking just telling users to install `python-backports.functools-lru-cache` with the system package manager might be the way to go until we officially drop Python 2 support.

I agree that having them install that via the system package manager is the right way to do things. I'd like it if the `--ensure-access` script could detect this condition and tell users what to do. :)

yuvadm, 2019-08-20:
@Qyriad @ktemkin To reiterate my comment from https://github.com/greatscottgadgets/libgreat/pull/5#issuecomment-513448744, some distros such as Arch, and possibly others, *do not* have that package to install. There's no reason for such a package to exist for Python 3-based installations.

ktemkin, 2019-08-21:
> @Qyriad @ktemkin To reiterate my comment from [greatscottgadgets/libgreat#5 (comment)](https://github.com/greatscottgadgets/libgreat/pull/5#issuecomment-513448744), some distros such as Arch, and possibly others, _do not_ have that package to install. There's no reason for such a package to exist for Python 3-based installations.

1. This issue specifically is with respect to using `python2`; which is unfortunately still necessary for a few key tools. In particular, the stable branch of `gnuradio` still requires py2, even on Arch. (For reference, Arch is my primary distribution, and has been for nearly fifteen years).

For those cases, Arch does indeed have a package to be installed:

```
extra/python2-backports.functools_lru_cache 1.5-1
    Backport of functools.lru_cache from Python 3.3
```

2. I'm a bit *less* concerned about detecting the case and providing a message for Arch users -- the "Arch Way" generally has users take a more active role in the management / hygiene of their package installations. Still, detecting the mixed-path case and providing an informational message seems like a nice courtesy, even to experienced users.

    I *am* concerned about users of distributions like Debian, Ubuntu, and Kali; and in general about users who are not incredibly familiar with Linux or their distro's package management. Many of our users install Linux in order to more easily run certain tools, and don't have (or *need*) the knowledge to figure out the solutions to complex package management situations like this one.

    The `ensure-access` script is designed *entirely* to help these users -- it'll help them get the tools they're interested in up and running quickly, without requiring them to undergo the cognitive overhead of learning about python and distribution package management. I see absolutely no reason *not* to provide them with suggestion that solves their problem.

    (I also firmly believe that users *should* be able to choose to install GreatFET via `pip`, or however they'd prefer. It's extremely important to me that a sense of 'technical cleanness' not create barriers to entry.)

3) Systems running on Arch, if managed per Arch standards, won't run into the mixed-path issue. Anyone creating an AUR package for GreatFET on py2 can include the relevant `extra` package in their `PKGBUILD`. :)

The issue of whether it's worth avoiding use of the `backports` module on py3 can be discussed further in your pull request, if you'd like. Either way, it's not the solution to _this_ issue.

yuvadm, 2019-08-21:
@ktemkin thanks for the thorough reply, I fully appreciate and agree with every single point you've made.

The only gripe I have is that this issue seems to be a duplicate of https://github.com/greatscottgadgets/libgreat/issues/2 which *is* a `python3` issue.

I might be missing something, but it's not clear to me how as an Arch user (or packager for that matter) I can do a plain `python3`-based installation without applying a patch similar to my proposal in https://github.com/greatscottgadgets/libgreat/pull/5

closed 2019-09-03 by ktemkin

xs2rahulz, 2020-06-05:
▪	This is again a python 3 issue , or multiple pythons in your system issue
	▪	Clean your dev machine
	▪	Remove python3
	▪	Make softlinks to go to python2
	▪	If configparser gives issue , remove it via sudo sudo pip uninstall configparser
	▪	And then install via user 

Similarly whatever module gives error , its because it either is still python3 redirected or via sudo
Easiest way is uninstall via sudo and install on user , DONT use ROOT

sudo pip uninstall backports.functools-lru-cache
pip install backports.functools-lru-cache