greatscottgadgets/greatfet #257: greatfet_msp430.py lacks Python2 support

mossmann, 2019-06-19:
```
$ python2 ./greatfet_msp430.py -I
Error, misidentified as 0x89.
Check wiring, as this should be 0x89 or 0x91.
```
This is just the first of various Python2 incompatibilities.

ikarus23, 2019-08-12:
Don't get me wrong. I admire the goal to make GreatFET python 2 compatible. But what is about https://pythonclock.org/? Python 2 will be out of support in 4 moth. Is the effort really worth it? Do I miss something?

ktemkin, 2019-08-22:
Our unofficial python2 support is mostly out of a need to be usable from a few key packages (e.g. `gnuradio`) that only support py2. :(

Once gnuradio has stable releases based on py3 (and they're available in distro repos), I'll more than gladly drop support for py2. :)

ikarus23, 2019-08-23:
Thanks Kate, I understand. Lets hope it will be soon. My Arch Linux already has gnuradio 3.8.0.0 since they released it 2 weeks ago. ;)

mossmann, 2023-02-16:
closing because we no longer try to support Python2

closed 2023-02-16 by mossmann