greatscottgadgets/greatfet #259: dfu-util: unable to read DFU status after completion

szczys, 2019-06-25:
I was playing around with Python apis and used onboard_flash which I believe overwrote the firmware? Trying now to use DFU to recover but not having success. Any idea where I'm going wrong?

```$ make greatfet_usb-program
[  7%] Built target libopencm3
[ 13%] Built target libgreat_module_bsp
[ 22%] Built target libgreat
[ 23%] Built target libgreat_module_debug-backtrace
[ 27%] Built target libgreat_module_comms
[ 28%] Built target libgreat_module_ad970x
[ 29%] Built target libgreat_module_allocator
[ 35%] Built target libgreat_module_usb
[ 37%] Built target libgreat_module_gpio
[ 37%] Built target libgreat_module_usb_comms
[ 79%] Built target greatfet_usb
[ 98%] Built target libgreatfet
[100%] Built target greatfet_usb.dfu.elf
[100%] Built target greatfet_usb.dfu.bin
[100%] Built target greatfet_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        83456 bytes
Download done.
dfu-util: unable to read DFU status after completion
[100%] Built target greatfet_usb-program
```

ktemkin, 2019-06-25:
> dfu-util: unable to read DFU status after completion 

This message is normal -- the DFU implementation on the NXP chips doesn't support status readback. Once you see the "download done" message, it's still programmed the board; `dfu-util` just emits an (admittedly error-looking) status message.

At that point, you should see the normal heartbeat on the GreatFET -- unless you've run into another issue. (I have seen reports that some Ubuntu distributions produce invalid binaries when building for DFU, which I haven't gotten a chance to look into).

Either way, if you have GreatFET installed from pip, you should be able to recover the original firmware by running this command when the board is in DFU mode:

```sh
gf fw --dfu --autoflash
```

Adding the `--dfu` argument should make `gf fw` arguments target DFU mode.

szczys, 2019-06-25:
Thanks! I wasn't getting a heartbeat after programming. The gf fw dfu method worked great and I'm back in business.

closed 2019-06-25 by szczys