greatscottgadgets/greatfet #260: Unable to flash GF1

kopasiak, 2019-07-12:
I've soldered two GF1 boards but I'm unable to flash or program any of those.
When I run:

```sh
 make greatfet_usb-program
```

The command finishes successfully but there the board performs a reset and there is no blinking LED nor USB connectivity

I tried to also flash the board but then I get:

```sh
/bin/sh: 1: greatfet_firmware: not found
```
I expect that this should be some symlink but I've no idea to which file it should point?

Any ideas how to fix flashing and what I could possibly debug to see what's wrong? Or could you please at least confirm that you are able to build and flash the firmware from current master branch and it blinks the LED after upload?

ktemkin, 2019-07-12:
Hi, Krzysztof:

You'll need to have the GreatFET host tools installed (or in your PYTHONPATH) for those commands to work. The easiest way to install is from pip:

```
$ pip install greatfet --user
```

If you want, once you have the tools installed, you can install the latest release firmware without building anything. With the device in DFU mode:

```
$ gf fw -d --autoflash
```

-- Kate

ktemkin, 2019-07-12:
Closing this, as it has a solution -- if the above doesn't work for you, feel free to reply and I'll re-open the issue. :)

closed 2019-07-12 by ktemkin

kopasiak, 2019-07-13:
Yeah, I had this installed, just forgot that sudo doesn't play well with python virtual env;)

Unfortunately this didn't took me too much further as the board is still not responding after:

```sh
 make greatfet_usb-program
```

There is no blinking LED and the device is not detected by the host:

```sh
[ 3652.549454] usb 1-1: new high-speed USB device number 31 using xhci_hcd
[ 3652.698426] usb 1-1: New USB device found, idVendor=1fc9, idProduct=000c
[ 3652.698429] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3652.698431] usb 1-1: Product: LPC
[ 3652.698432] usb 1-1: Manufacturer: NXP
[ 3652.698433] usb 1-1: SerialNumber: ABCD
[ 3665.824823] usb 1-1: USB disconnect, device number 31
```

I've tried also to flash the blinky firmware but effect was pretty much the same:(

Do you have any ideas what I could check before I get my JTAG? Is there some serial console or sth?

BTW. @ktemkin 
Could you please reopen issue as I'm unable to do so?

kopasiak, 2019-07-13:
Fast update from my site. I was able to get the board working using the command that you mentioned:

```sh
gf fw -d --autoflash
```

I guess that this command flashes the prebuild firmware to the the board so it's not actually what I wanted to achieve but at least I know that the HW is OK. I have seen in one of the issues that you are aware of build problems on Ubuntu so maybe it's just one of them? I'm currently running Ubuntu 18.04. Could you please share on which distro you are building the firmware and which version of tools are you using?