greatscottgadgets/greatfet #261: Unable to build firmware on Ubuntu 18.04

kopasiak, 2019-07-17:
I'm trying to build greatfet firmware on Ubuntu 18.04. Compilation goes smoothly, there is no outstanding errors and I get two output files:
greatfet_usb.bin
greatfet_usb_dfu

The issue is that if I try to upload greatfet_usb to RAM using DFU the process finishes but there is no heartbeat nor USB connectivity to the board.

On the other hand, if I first preflash the board with flash_stub.dfu that comes with host tools python package, then I can successfully upload greatfet_usb.bin to the flash and after reset everything is working fine. 

Any ideas what could be wrong? I've already changed compiler as suggested in #239

kopasiak, 2019-07-17:
I've just also tried to build flash_stub but effect is the same. I've attached a flash_stub binary built on my system for you to test. I'm building from latest release tag which is v2019.5.1.dev0.
[flash_stub.zip](https://github.com/greatscottgadgets/greatfet/files/3403858/flash_stub.zip)

coley7, 2019-08-09:
I figure out similar problems.

For me also only works when using the python package which comes from pip.
DFU works here, but when it flashs to the SPI flash and resets nothing happens again...

@kopasiak what have you done to get it work?

With the binaries from the git repo it doesn't work...

kopasiak, 2019-08-09:
The answer is quite straight forward. I've never managed to get it to work but I've found some workaround.

Basically there are two parts of the firmware that you are building the flash-stub and the main firmware. Even through you should be able to flash the main firmware via dfu the flashing process is usually done two-steps. First you flash the stub via dfu and then you flash the main firmware using the loaded flash-stub.

It turns out that only the part that you flash via dfu is broken so what I did is I first flash the flash-stub that comes prebuild with python package and then I flash my custom-build firmware

ktemkin, 2019-08-09:
Are you, by chance, on a Debian-derived distribution (e.g. Debian, Ubuntu, Kali)? 

I've seen an issue where their compilers seem to produce invalid DFU binaries -- though the non-DFU binaries seem to work fine. I haven't had a chance to look into it.

kopasiak, 2019-08-09:
Yes I do, I'm on Ubuntu 18.04.

I've seen this ticket and tried even to use some old compiler as suggested somewhere but it didn't helped.

coley7, 2019-08-10:
so I tried it on a Debian(4.9) and Ubuntu (18.04) 

```
$:~/greatfet/firmware/build/greatfet_usb$ greatfet_firmware -w greatfet_usb.bin  
Trying to find a GreatFET device...
libgreat compatible board in flash-stub mode found. (Serial number: 0000212200090041a000)  
Writing data to SPI flash...  
Written 83736 bytes of 83736.  
Write complete!  
Reset not specified; new firmware will not start until next reset.
```

When I do now the reset, no heartbeat comes, everything is lost
I did ``gf fw --dfu --autoflash`` from a vm and flashed dfu. That worked. The firmware was also flashed but it didn't work after reset.
and I also tried using the first step of ``gf fw --dfu --autoflash`` and the command ``greatfet_firmware -w greatfet_usb.bin`` with the compiled output provided in the git repo but that didn't work, too.

I also say, that I dont't have X2 connected, but only X1. But I think that should make no difference...

ktemkin, 2019-08-24:
So -- @qyriad and I spent some time looking at this earlier this week, and it turned out the issue was *cmake* related. Our CMake definitions were accidentally applying the wrong linker script due to a change in where it keeps a linker-related option.

That said, the whole process inspired me to restructure how we handle upload-to-RAM in DFU mode. Current master now has runtime relocation detection, and uses [pyfwup](https://github.com/hacking-usb/pyfwup) instead of `dfu-util`. The practical upshot of this is that we can use the same `.bin` files for flashing-to-SPI and for DFU upload, so this issue is now moot. 

Things should "just work" on the latest master -- so I'm going to mark this issue as solved. If anyone wants to validate that that's the case. I'd appreciate comments here. If it turns out things are still broken for anyone, we can re-open the issue and continue debugging. :)

closed 2019-08-24 by ktemkin

kopasiak, 2019-08-29:
I've just checked the recent master branch and now it all works fine for me. Thank you @ktemkin and @Qyriad  for fixing this issue.

coley7, 2019-09-14:
Thank you,
for me the same issue as described above exists still. After dfu I see the heartbeat, after SPI flashing nothing goes on...

ktemkin, 2019-09-14:
@coley7 To confirm, you're saying that _DFU_ works, but writing to _flash_ doesn't? That'd be a different behavior than what others have reported in the issue.

coley7, 2019-09-15:
DFU works, It writes also to flash (the program says it writes to flash) but after reset nothing appears

kopasiak, 2019-09-15:
Make sure that you rebuild all the components including dependencies. It would be best if you could try a fresh clone a do a full rebuild step by step as cmake likes to have some leftovers (unless you rm -rf all build directories, that's also a valid option)

coley7, 2019-09-16:
ok I have missed the installation... 
```
debian@debianvmws:~/greatfet/firmware/build/greatfet_usb$ greatfet_firmware -w greatfet_usb.bin
Trying to find a GreatFET device...
libgreat compatible board in flash-stub mode found. (Serial number: 000075b463d92644aea3)
Writing data to SPI flash...

Traceback (most recent call last):                                              
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-b_v2019.9.1_.dev_git.b_3c67a817_-py3.5.egg/greatfet/programmers/firmware.py", line 76, in write
    self.comms.get_exclusive_access()
AttributeError: 'USBCommsBackend' object has no attribute 'get_exclusive_access'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/greatfet_firmware", line 11, in <module>
    load_entry_point('GreatFET===b-v2019.9.1-.dev-git.b-3c67a817-', 'console_scripts', 'greatfet_firmware')()
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-b_v2019.9.1_.dev_git.b_3c67a817_-py3.5.egg/greatfet/commands/greatfet_firmware.py", line 214, in main
    spi_flash_write(device, args.write, args.address, log_function)
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-b_v2019.9.1_.dev_git.b_3c67a817_-py3.5.egg/greatfet/commands/greatfet_firmware.py", line 57, in spi_flash_write
    progress_callback=lambda offset,  len : progress.update(page_size))
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-b_v2019.9.1_.dev_git.b_3c67a817_-py3.5.egg/greatfet/programmers/firmware.py", line 282, in upload
    self.write(data, address, erase_first, progress_callback)
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-b_v2019.9.1_.dev_git.b_3c67a817_-py3.5.egg/greatfet/programmers/firmware.py", line 79, in write
    self.comms.release_exclusive_access()
AttributeError: 'USBCommsBackend' object has no attribute 'release_exclusive_access'

```
What could be the problem there?

ktemkin, 2019-09-16:
Sounds like you upgraded GreatFET without upgrading pygreat. Make sure you’ve installed a dev version of pygreat as well (from libgreat/host). :)

coley7, 2019-09-16:
thanks, I also did ìn libgreat/host ` sudo python3 setup.py install `.
Has it be a special commit? I was working on the tag v2019.9.1

Tried also with python2 but then another error occours:

```
debian@debianvmws:~/greatfet$ gf fw -d --autoflash
Traceback (most recent call last):
  File "/usr/local/bin/greatfet_fw", line 11, in <module>
    load_entry_point('GreatFET==2019.9.1.dev0+git.e254b321.dirty', 'console_scripts', 'greatfet_fw')()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2843, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2434, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2440, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/GreatFET-2019.9.1.dev0+git.e254b321.dirty-py2.7.egg/greatfet/commands/greatfet_firmware.py", line 23, in <module>
    from greatfet.utils import log_silent, GreatFETArgumentParser
  File "/usr/local/lib/python2.7/dist-packages/GreatFET-2019.9.1.dev0+git.e254b321.dirty-py2.7.egg/greatfet/utils.py", line 19, in <module>
    from greatfet.boards.flash_stub import GreatFETFlashStub
ImportError: No module named boards.flash_stub

```

Qyriad, 2019-09-17:
Looks like `greatfet` was installed with Python 2, but `libgreat` with Python 3. I recommend `pip2 uninstall greatfet` followed by `pip3 install greatfet` or `python3 setup.py install`.

coley7, 2019-09-18:
thanks. That worked in addition to remove all relevant packages in the dist-packes folder. 
Now the flash process worked :-)

```
debian@debianvmws:~/greatfet/host$ gf fw -d --autoflash
Trying to find a GreatFET device...
libgreat compatible board in flash-stub mode found. (Serial number: 000075b463d92644aea3)
Writing data to SPI flash...

                                                                                
Write complete!
Resetting GreatFET...
Reset complete!
```

The problem is now as before, that nothing happens after the reset. No LED, no USB device appears... 
Any idea? How to find out what's the problem? I have no debugger for the board available.

coley7, 2019-09-20:
also nothing happens...

I tried following ones:
greatfet-2019-09-17-build_506-git_3572a24.tar.xz
greatfet-2019-09-17-build_505-git_c91bda3.tar.xz
greatfet-2019-09-13-build_499-git_67c6421.tar.xz
greatfet-2019-09-02-build_479-git_9a2c430.tar.xz

Is there a way to verify that it's not a hardware issue?

mrstock, 2019-09-20:
did you try ? :

`gf info`

and

`gf dmesg`

coley7, 2019-09-20:
thanks for your response. I did `gf fw -d --autoflash` and executed the following commands after the DFU flashing.

```

debian@debianvmws:~$ gf info
Found a libgreat compatible board in flash-stub mode!
  Board ID: 10000
  Firmware version: git-v2019.9.1-1-ge254b32-dirty
  Part ID: a0000a304f4769
  Serial number: 000075b463d92644aea3
 

debian@debianvmws:~$ gf dmesg
[------------] System started after hard reset / power cycle.
[       93813] System clock bringup complete.
[       93827] Configuring board pins...
[       93876] Starting in flash stub mode.
[       93887] Awaiting firmware programming commands.
)[    11832522] comms error: requested a USB response when no communications were underway! (stage: 0)
[    22371408] comms error: requested a USB response when no communications were underway! (stage: 0)

```

mrstock, 2019-09-20:
hum ... your board is in flash-stub mode ...

A normal behavior should be : 

```
Stock:mrstockmac ~ > gf info
Found a GreatFET One!
  Board ID: 0
  Firmware version: git-v2019.9.1-14-g3572a24
  Part ID: a0000a30714f6a
  Serial number: 000055cc66e334454f23
```
```
Stock:mrstockmac ~ > gf dmesg
[------------] System started after hard reset / power cycle.
[       88754] System clock bringup complete.
[       88786] Configuring board pins...
[       88837] GreatFET initialization complete!
```

coley7, 2019-09-24:
The -d option is afaik for DFU. What I want to say is that the board is only visible as DFU device for me. Afterwards the SPI chip is flashed and the reset comes nothing is happening...
So `gw fw --auto` is not working for me because I need the DFU mode.

mrstock, 2019-09-24:
I see, sorry I can't help you on this issue, maybe @ktemkin or @Qyriad can tell you something more.

ktemkin, 2019-09-24:
> Edit : can you try with ? :
> 
> `gf fw --auto`
> ( avoiding the -d option )

This is a good suggestion. Can you try this immediately after issuing your DFU command (while your board is still in flash-stub mode), and report the output? So:

```
gf fw -d --autoflash
gf fw --autoflash
```

Normally, the second command wouldn't work -- but it looks like your board is still leaving things in `flash-stub` mode.

Can you also try:

```
gf fw -V --autoflash
```

and see if your GreatFET boots?

coley7, 2019-11-01:
sorry for the long delay. Same issue as before.
Is there a way to verify that's not a problem with the hardware?