greatscottgadgets/greatfet #282: No backend available on macOS Mojave ( 10.14.6 )

mrstock, 2019-09-09:
I'm trying to run greatfet on a mac without success :

```
mac ~ > gf info

Traceback (most recent call last):
  File "/usr/local/bin/greatfet_info", line 10, in <module>
    sys.exit(main())
  File "/Users/Stock/Library/Python/3.7/lib/python/site-packages/greatfet/commands/greatfet_info.py", line 94, in main
    devices = GreatFET(find_all=True)
  File "/Users/Stock/Library/Python/3.7/lib/python/site-packages/greatfet/greatfet.py", line 31, in GreatFET
    return GreatFETBoard.autodetect_all(**board_identifiers)
  File "/usr/local/lib/python3.7/site-packages/pygreat/board.py", line 105, in autodetect_all
    subclass_devices = subclass.all_accepted_devices(**device_identifiers)
  File "/usr/local/lib/python3.7/site-packages/pygreat/board.py", line 136, in all_accepted_devices
    raw_devices = usb.core.find(**identifiers)
  File "/usr/local/lib/python3.7/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

Any attempt to use the usb port produces this effect.

I have already installed the following packages :

pip3 install pyusb
pip3 install libusb

Someone managed to get it running ? or could you suggest me a way ?

Thank you

Qyriad, 2019-09-09:
You'll have to install the libusb C library; the libusb pip package isn't necessary. If you have Homebrew, you should be able to run `brew install libusb` to take care of that.

mrstock, 2019-09-10:
```
brew install libusb
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
baobab               folly                librsvg              pspg                 sccache              vulkan-headers
file-roller          glooctl              pdftk-java           rustup-init          virgil               you-get

Warning: libusb 1.0.22 is already installed and up-to-date
To reinstall 1.0.22, run `brew reinstall libusb`
```
Thank you, but same result

mrstock, 2019-09-10:
I found a solution : 

macOS search for the library "libusb" through this code : 

`https://github.com/python/cpython/blob/master/Lib/ctypes/util.py`

```
elif os.name == "posix" and sys.platform == "darwin":
    from ctypes.macholib.dyld import dyld_find as _dyld_find
    def find_library(name):
        possible = ['lib%s.dylib' % name,
                    '%s.dylib' % name,
                    '%s.framework/%s' % (name, name)]
        for name in possible:
            try:
                return _dyld_find(name)
            except ValueError:
                continue
        return None
```

The backend libusb1 search for this terms : 

```
return usb.libloader.load_locate_library(
                ('usb-1.0', 'libusb-1.0', 'usb'),
                'cygusb-1.0.dll', 'Libusb 1',
                win_cls=win_cls,
                find_library=find_library, check_symbols=('libusb_init',))
```
so the name libusb-1.0.0.dylib will never be found.

Workaround : create a symlink named **libusb-1.0.dylib** linked to the right library **libusb-1.0.0.dylib**

```
 rwxr-xr-x   1 Stock  admin    99880 10 Set 18:49 libusb-1.0.0.dylib*
lrwxr-xr-x   1 Stock  admin       18 10 Set 19:28 libusb-1.0.dylib@ -> libusb-1.0.0.dylib
```

Qyriad, 2021-02-24:
It looks like the upstream bug has been fixed. Closing. If anyone is still having this issue I'll reopen.

closed 2021-02-24 by Qyriad