greatscottgadgets/greatfet #316: undefined reference to `capture_buffer_write_position`

cdwilson, 2019-11-17:
I'm trying to build the firmware for the first time, and when I run `make`, I'm getting the linker error below.  Any idea what may be causing this?  (I just got a GreatFET last night, so I'm not sure if I missed something, or if this is an actual issue)

In case it matters, I'm trying to compile using the GNU Arm Embedded Toolchain
Version `9-2019-q4-major` installed via homebrew on macOS 10.15.1 (https://github.com/cdwilson/homebrew-formulae/blob/eb26c7130b170472c48605f89adcd06cc300871c/arm-none-eabi-gcc.rb).  I also tried reverting to the previous `8-2019-q3-update` but I'm getting the same error.

```
[100%] Linking ASM executable rhododendron_m0.bin.elf
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/rhododendron_m0.bin.elf.dir/rhododendron/rhododendron_m0.S.obj: in function `fault_handler':
/Users/chris/Projects/greatfet/firmware/greatfet_usb/rhododendron/rhododendron_m0.S:80: undefined reference to `capture_buffer_write_position'
collect2: error: ld returned 1 exit status
make[3]: *** [greatfet_usb/rhododendron_m0.bin.elf] Error 1
make[2]: *** [greatfet_usb/CMakeFiles/rhododendron_m0.bin.elf.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [firmware] Error 2
```

zhaohuansi, 2019-11-17:
I am the same problem.  KALI 2019

ktemkin, 2019-11-19:
This was a rebasey-issue, and should be fixed. CI build is passing, so I'm closing this.

Let me know if you have any issues, still. :)

closed 2019-11-19 by ktemkin

cdwilson, 2019-11-19:
Pulled the latest and it's working now.  Thanks Kate!