greatscottgadgets/greatfet #321: Issue building NXP LPC4330 Xplorer

decidedlygray, 2019-11-26:
Hello,

I'm hoping you can help me out. I'm currently not able to build the firmware for the NXP LPC4330 Xplorer board. I've read up on the issue here: #1  and followed the instructions here: https://github.com/greatscottgadgets/greatfet/wiki/Getting-Started-with-Firmware-Development

`libopencm3` built fine, but with `greatfet_usb` I get all the way here and get some errors about undeclared identifiers:
```
dg@iot:/opt/greatfet/firmware/greatfet_usb/build$ cmake -DBOARD=NXP_XPLORER ..
-- Fetching git submodules.
-- Building for NXP LPC4330 Xplorer.
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/lib/arm-none-eabi
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/greatfet/firmware/greatfet_usb/build
dg@iot:/opt/greatfet/firmware/greatfet_usb/build$ make
[  6%] Built target libopencm3
[  7%] Built target libgreat_module_dac
[  9%] Built target libgreat_module_allocator
[ 12%] Built target libgreat_module_comms
[ 13%] Built target libgreat_module_ringbuffer
[ 13%] Built target libgreat_module_ad970x
[ 13%] Built target libgreat_module_scheduler
[ 14%] Built target libgreat_module_debug-backtrace
[ 19%] Built target libgreat_module_usb
[ 20%] Built target libgreat_module_gpio
[ 20%] Built target libgreat_module_usb_comms
[ 23%] Built target libgreat_module_sgpio
[ 24%] Built target libgreat_module_uart
[ 69%] Built target greatfet_usb
[ 75%] Built target libgreat_module_bsp
[ 84%] Built target libgreat
[ 85%] Building C object common/CMakeFiles/libgreatfet.dir/greatfet_core.c.obj
/opt/greatfet/firmware/common/greatfet_core.c: In function 'pin_setup':
/opt/greatfet/firmware/common/greatfet_core.c:118:41: error: 'SCU_SSP1_SCK_FUNC' undeclared (first use in this function)
  scu_pinmux(SCU_SSP1_SCK,  SCU_SSP_IO | SCU_SSP1_SCK_FUNC);
                                         ^~~~~~~~~~~~~~~~~
/opt/greatfet/firmware/common/greatfet_core.c:118:41: note: each undeclared identifier is reported only once for each function it appears in
/opt/greatfet/firmware/common/greatfet_core.c:119:41: error: 'SCU_SSP1_MISO_FUNC' undeclared (first use in this function)
  scu_pinmux(SCU_SSP1_MISO, SCU_SSP_IO | SCU_SSP1_MISO_FUNC);
                                         ^~~~~~~~~~~~~~~~~~
/opt/greatfet/firmware/common/greatfet_core.c:120:41: error: 'SCU_SSP1_MOSI_FUNC' undeclared (first use in this function)
  scu_pinmux(SCU_SSP1_MOSI, SCU_SSP_IO | SCU_SSP1_MOSI_FUNC);
                                         ^~~~~~~~~~~~~~~~~~
/opt/greatfet/firmware/common/greatfet_core.c:121:41: error: 'SCU_SSP1_SSEL_FUNC' undeclared (first use in this function)
  scu_pinmux(SCU_SSP1_SSEL, SCU_SSP_IO | SCU_SSP1_SSEL_FUNC);
                                         ^~~~~~~~~~~~~~~~~~
common/CMakeFiles/libgreatfet.dir/build.make:302: recipe for target 'common/CMakeFiles/libgreatfet.dir/greatfet_core.c.obj' failed
make[2]: *** [common/CMakeFiles/libgreatfet.dir/greatfet_core.c.obj] Error 1
CMakeFiles/Makefile2:311: recipe for target 'common/CMakeFiles/libgreatfet.dir/all' failed
make[1]: *** [common/CMakeFiles/libgreatfet.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
dg@iot:/opt/greatfet/firmware/greatfet_usb/build$ 
```

Looks like its a problem related to this commit: https://github.com/greatscottgadgets/greatfet/commit/a41c61f23f8d53fa2f7eae4d9c8c131fe3827f80

decidedlygray, 2019-11-26:
I also tried using pre-libgreat, but then I run into this issue: https://github.com/greatscottgadgets/greatfet/issues/200

ktemkin, 2020-01-20:
I'd have to check this out -- since the LPC4330 Xplorer was discontinued, we haven't included it in our main tests. I can't promise it'll be soon (given the Xplorer is EOL'd), but I'll keep this on my TODO list.

straithe, 2021-11-13:
Do you still need assistance, @decidedlygray ?

decidedlygray, 2021-11-19:
Eh. It would be cool to build but the board has been discontinued for a few years now. I'll close this out. Thanks for following up!

closed 2021-11-19 by decidedlygray