greatscottgadgets/greatfet #333: command line i2c util failing on write

aloeppert, 2020-02-23:
Hello GreatFET community. I recently tried using the great fet out of the box to do some i2c raw commands so I can verify expected values and communication with a known good host.  However I'm not seeing the expected results using the i2c utility.  I've posted some relevant command output to show my revision and that the scan function works as expected (showing the device at 0x60).

$ gf info
Found a GreatFET One!
  Board ID: 0
  Firmware version: v2020.1.2
  Part ID: a0000a30604766
  Serial number: <redacted>
 
$ gf i2c -z
Working I2C address(es): (Address, RW bit, frequency)
0x60 W [100000, 400000]
0x60 R [100000, 400000]

******** W/R bit set at each valid address ********
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: WR -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

$ gf i2c -a 0x60 -w 0x0c -r 2 -v
Trying to find a GreatFET device...
GreatFET One found. (Serial number: <redacted>)
Writing to address 0x60
Traceback (most recent call last):
  File "/home/loeppert/.local/bin/greatfet_i2c", line 10, in <module>
    sys.exit(main())
  File "/home/loeppert/.local/lib/python3.7/site-packages/greatfet/commands/greatfet_i2c.py", line 48, in main
    write(device, args.address[0], args.write, log_function)
  File "/home/loeppert/.local/lib/python3.7/site-packages/greatfet/commands/greatfet_i2c.py", line 73, in write
    log_function("I2C write status: %s" % hex(i2c_status))
TypeError: 'tuple' object cannot be interpreted as an integer

closed 2020-02-23 by ktemkin

ktemkin, 2020-02-23:
Oops! Thanks for finding that -- that was a bug in the `i2c` command frontend.  It should be fixed in `master`. :)

Let me know if you need help getting onto e.g. a nightly build, or if you have any more trouble.

aloeppert, 2020-02-23:
Thanks for fixing so quickly. I'll probably not get to trying again until
tomorrow. Good to know there are people behind the project!

PR #334 (2020-02-23): switch to f-strings in greatfet_i2c, and don't print the status
Fixes #333.