greatscottgadgets/greatfet #336: Greatfet py module is not installed

abahmer, 2020-04-02:
HI, 
I have installed the py package by
pip3 install --upgrade --user greatfet
and set path such that I can 

**(base) ab@ab-Aspire-A114-32:~/anaconda3/envs/facedancer_new/Facedancer/examples$ gf info
Found a GreatFET One!
  Board ID: 0
  Firmware version: v2020.1.2
  Part ID: a0000a306d435c
  Serial number: 000057cc67e6315f1657**
 
but when I run ./rubber-ducky.py, it misses the module. What is wrong? Thanks!

**"INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
Traceback (most recent call last):
  File "./rubber-ducky.py", line 37, in <module>    main(device, type_letters())
  File "/home/ab/anaconda3/lib/python3.7/site-packages/facedancer-2.9-py3.7.egg/facedancer/devices/__init__.py", line 51, in default_main
  File "/home/ab/anaconda3/lib/python3.7/site-packages/facedancer-2.9-py3.7.egg/facedancer/future/device.py", line 172, in emulate
  File "/home/ab/anaconda3/lib/python3.7/site-packages/facedancer-2.9-py3.7.egg/facedancer/future/device.py", line 124, in connect
  File "/home/ab/anaconda3/lib/python3.7/site-packages/facedancer-2.9-py3.7.egg/facedancer/core.py", line 23, in FacedancerUSBApp
  File "/home/ab/anaconda3/lib/python3.7/site-packages/facedancer-2.9-py3.7.egg/facedancer/core.py", line 56, in autodetect
facedancer.errors.DeviceNotFoundError
sys:1: RuntimeWarning: coroutine 'type_letters' was never awaited**

ktemkin, 2020-04-07:
Are you sure the two installs are making it into the same environment? It looks like your FaceDancer install is running in Anaconda; but you may be using your system pip to install GreatFET. :)

abahmer, 2020-04-07:
Thank you! I have solved the problem! Andreas

 
 

Gesendet: Dienstag, 07. April 2020 um 18:37 Uhr

closed 2020-06-01 by ktemkin