greatscottgadgets/greatfet #341: Change SPI peripheral clock frequency?

rflmota, 2020-07-11:
Hello, I'm a recent GreatFET user and not much proficient in Python, and I've been struggling to find a way to change the default SPI peripheral clock frequency with GreatFET API. I've looked in the docs, search through code-base and GitHub issues but wasn't able to provide any examples on how to accomplish that. Can somebody provide me some hints? ty

mossmann, 2020-07-12:
I believe that you should be able to set the SPI clock frequency by instantiating a new SPIBus, like this:
```python
from greatfet.interfaces.spi_bus import SPIBus
cs = gf.gpio.get_pin('J1_P4')
spi_bus=SPIBus(gf, chip_select_gpio=cs, freq_preset=SPIBus.FREQ.C51000000Hz)
spi_bus.transmit((0,))
```
However, this does not appear to be working. The clock frequency remains the same regardless of the freq_preset or other clock arguments I have tried. This may be a bug.

rflmota, 2020-07-12:
Hello @mossmann, indeed that's exactly what I've tried with the same result, so I was wondering if I wasn't issuing the right commands. I've even tried with a sequence of enable/disable peripheral and outside of "gf shell" prompt to make sure this wasn't a setting only able to configure at first instantiation, but the result was the same. I'll be following updates to a possible bug fix.

Thank you for your attention.

PR #342 (2020-07-17): firmware: fix SPI frequency setting
Fixes #341.
closed 2020-07-17 by ktemkin