greatscottgadgets/greatfet #344: max_packet_size_ep0 < 32bytess does not work

mrh1997, 2020-08-02:
When simulating a device with a max_packet_size_ep0 of 8 or 16 Bytes (32/64 Bytes work) replying to the 4th request will fail. In fact I do not know if it is definitiely the 4th request, but I did not find any other commonality (like number of transactions or timing).

I tested on Windows where the 3rd request is "GetDescriptor CONFIGURATION" which is processed correctly. The 4th request ("GetDescriptor STRING iSerialNumber") is received correctly and FaceDancer transmits the correct answer via [GreatDancerApp.send_on_endpoint()](https://github.com/mrh1997/Facedancer/blob/e1095030aa6dce764bc03c2f99a13473e3269de6/facedancer/backends/greatdancer.py#L246). 

But the firmware does not transfer the answer (any NAKs are transferred) while ``GreatDancerApp.send_on_endpoint()`` returns without error.

**important:** To make FaceDancer work  with max_packet_size_ep0 < 64 [this buggy line](https://github.com/mrh1997/Facedancer/blob/master/facedancer/future/device.py#L126-L126) has to be fixed to ``self.backend.connect(self, self.max_packet_size_ep0)``

mrh1997, 2020-08-03:
The following FaceDancer code can be used to reproduce the bug.

```!python
from facedancer         import main
from facedancer.future  import *

@use_inner_classes_automatically
class BugDemoDevice(USBDevice):
    max_packet_size_ep0      : int  = 8
    class Configuration1(USBConfiguration):
        configuration_number   : int = 1

if __name__ == '__main__':
    main(BugDemoDevice)
```

Please note that [this FaceDancer pullrequest](https://github.com/usb-tools/Facedancer/pull/45) is required to reproduce the bug

The outcome you can see in the analyser is like:

[image]

mrh1997, 2020-08-03:
I forgotten to mention, than when setting ``max_packet_size_ep0      : int  = 64`` everything works as expected (aka GetDescriptor STRING and following commands are working)

ktemkin, 2020-08-10:
Hey! Thanks for the report!

@Qyriad would you be up for taking a look at this?

mrh1997, 2020-09-28:
Did some further investigations by now:
- the value of ``max_packet_size_ep0`` is [forwarded correctly to the LPC's USB controller](https://github.com/greatscottgadgets/libgreat/blob/e8351008da46f23c875c0207accdfd2ca2d2ed79/firmware/platform/lpc43xx/drivers/usb/usb.c#L868-L868) (This can also be seen in the log above, as the transfers are split correctly into 8 byte transactions)
- Incidentially the serialnumber was too big to fit into a single 16 byte transfer but it fitted into a single 32 byte transfer. Thus I testwise added the line ``serial_number_string: str = "12"`` to my testscript. Now the answer to "GetDescriptor (String 3)" fitted into a single (8 byte) transaction. But even then the problem occured...

mrh1997, 2020-09-30:
The Problem is, that the USB_TD_DTD_TOKEN_STATUS_ACTIVE will not be cleared if the ``max_packet_size`` of enpoint 0 is 8 or 16. This causes [usb_queue_clean_up_transfers()](https://github.com/greatscottgadgets/libgreat/blob/e8351008da46f23c875c0207accdfd2ca2d2ed79/firmware/platform/lpc43xx/drivers/usb/usb_queue.c#L199) not to to free the dTD buffers after a IN transfer on endpoint 0 is completed. As [there are only 4 buffers for PIPE 0 IN](https://github.com/greatscottgadgets/greatfet/blob/master/firmware/greatfet_usb/usb_endpoint.c#L30-L30) it is not possible to [allocate](https://github.com/greatscottgadgets/libgreat/blob/e8351008da46f23c875c0207accdfd2ca2d2ed79/firmware/platform/lpc43xx/drivers/usb/usb_queue.c#L114-L114) memory for a further transfers to be send.

I assume this is a bug in the LPC hardware because:
- As mentioned above a ``max_packet_size`` of 32 and 64 works fine.
- I doublechecked the LPC datasheet. Looks like the code is OK.

 @Qyriad: any idea for a workaround?

ktemkin, 2020-09-30:
This might be a case for expiring packets before they're marked non-active. They're currently at least expired and cleaned up when the next SETUP comes through; but perhaps there's another condition we could use for expiry?

Do we know if the interrupt-on-complete still fires?

mrh1997, 2020-09-30:
Actually the ``usb.c`` Module [flushes/frees already all transfers on endpoint 0](https://github.com/greatscottgadgets/libgreat/blob/e8351008da46f23c875c0207accdfd2ca2d2ed79/firmware/platform/lpc43xx/drivers/usb/usb.c#L963-L963) when receiving a SETUP packet.

But I saw that ``greatdancer.c`` does only rely partially on ``usb.c`` and implements a lot of the lowlevel logic on its own. Especially interrupt handling (including the flushing of endpoint 0 on SETUP packet) is not reused by ``greatdancer.c``. Thus I would add flushing to the [greatdancer_verb_read_setup()](https://github.com/greatscottgadgets/greatfet/blob/master/firmware/greatfet_usb/classes/greatdancer.c#L352-L352).

Hope you are OK with that...

Qyriad, 2020-10-01:
Fixed with #353.

closed 2020-10-01 by Qyriad