greatscottgadgets/greatfet #349: Unable to sync greatfet firmware with gf fw --auto on kali

pickeditmate, 2020-08-31:
I am attempting to get Jakes fork on to the greatfet but running in to a problem syncing with the host tools when running "gf fw --auto" which produces an error of "unrecognized argument --auto"

I believe the host tools are set up correctly as per greatfet/host/README.md

Please advise

Here is the output of gf info -H

kali@kali:~/JakeGraves/greatfet/host/greatfet$ gf info -H
Host tools info:
        host module version: b-v2020.1.2-.dev-git.b-f1143fa2-
        pygreat module version: 2020.1.2
        python version: 3.8.2 (default, Apr  1 2020, 10:45:58)

        module path: /usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_f1143fa2_-py3.8.egg/greatfet
        command path: /usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_f1143fa2_-py3.8.egg/greatfet/commands
        gnuradio-companion block path: /usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_f1143fa2_-py3.8.egg/greatfet/gnuradio

Found a GreatFET One!
  Board ID: 0
  Firmware version: v2020.1.2
  Part ID: a0000a306d4f6a
  Serial number: 000057cc67e632725257

GravesJake, 2020-08-31:
`--autoflash` is meant for when you [install the host tools via pip3](https://greatscottgadgets.github.io/greatfet-tutorials/getting-started.html). When building host/firmware from source, I'd recommend following the directions found [here](https://github.com/greatscottgadgets/greatfet/wiki/Getting-Started-with-Firmware-Development).

pickeditmate, 2020-08-31:
Thank you Jake, I've just git cloned your chipcon branch so will let you know how it goes on the IM-ME in the next few days.

GravesJake, 2020-09-01:
Sweet :)

closed 2020-09-01 by GravesJake

pickeditmate, 2020-09-01:
Try as I may, I cannot get the firmware to build on kali by either using python or python3. Is the bin file uploaded to anywhere so I can test with Jake's chipcon build?

GravesJake, 2020-09-01:
From `/greatfet/`:
```
cd firmware/libopencm3
make clean
make
cd ../greatfet_usb
mkdir build
cd build
cmake ..
make clean
make
```
Doesn't produce a `greatfet_usb.bin` file?

pickeditmate, 2020-09-01:
I get the following after building with python 2.7

kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$ sudo mkdir build
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$ cd build
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$ cmake ..
-- Fetching git submodules.
error: could not lock config file /home/kali/JakeGraves/greatfet/.git/modules/firmware/libopencm3/config: Permission denied
fatal: could not set 'core.worktree' to '../../../../firmware/libopencm3'
CMake Error at /home/kali/JakeGraves/greatfet/firmware/cmake/submodules.cmake:25 (message):
  Pulling down submodules failed! (1)
Call Stack (most recent call first):
  /home/kali/JakeGraves/greatfet/firmware/cmake/greatfet_prelude.cmake:48 (include)
  CMakeLists.txt:10 (include)

-- Configuring incomplete, errors occurred!
CMake Error: Cannot open file for write: /home/kali/JakeGraves/greatfet/firmware/greatfet_usb/build/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /home/kali/JakeGraves/greatfet/firmware/greatfet_usb/build/CMakeCache.txt
CMake Error: : System Error: Permission denied
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$ sudo cmake ..
-- Fetching git submodules.
-- Building for GreatFET Azalea.
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/lib/arm-none-eabi
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$ sudo make clean
make: *** No rule to make target 'clean'.  Stop.
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$ sudo make
make: *** No targets specified and no makefile found.  Stop.
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$

GravesJake, 2020-09-01:
What happens if you skip `make clean`?

pickeditmate, 2020-09-01:
Sorry for the delay, I had to deal with something at home.

If I go straight for sudo make I get this

kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$ make
make: *** No targets specified and no makefile found.  Stop.
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$ sudo make
[sudo] password for kali:
make: *** No targets specified and no makefile found.  Stop.
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$

GravesJake, 2020-09-01:
Try rerunning `cmake ..` as root.

pickeditmate, 2020-09-01:
It's not looking good still. I can paste my full output to pastebin if that may help. But here is the error 

kali@kali:~/JakeGraves/greatfet/firmware/libopencm3$ cd ../greatfet_usb
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$ sudo mkdir build
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb$ cd build
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$ sudo cmake ..
-- Fetching git submodules.
-- Building for GreatFET Azalea.
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/lib/arm-none-eabi
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$ sudo make
make: *** No targets specified and no makefile found.  Stop.
kali@kali:~/JakeGraves/greatfet/firmware/greatfet_usb/build$

GravesJake, 2020-09-01:
What version of cmake do you have installed?

GravesJake, 2020-09-01:
Also, nightly builds are available here: https://nightlies.ktemkin.com/

pickeditmate, 2020-09-01:
Thanks Jake, I'll give the nightlies a go but really wanted to play with your chipcon addition.

cmake is on v3.16.3-3

GravesJake, 2020-09-01:
The chipcon tool is still usable inside of `gf shell`, just not the command-line tool yet. But your cmake version looks to be compatible, my guess is that this is a Kali issue (those happen a lot for us). Could you try a different distro?

pickeditmate, 2020-09-01:
Yeah I'm happy to, are there any diatros you would suggest that can go straight on to a pi4? Tbh as long as I have the bin on the greatfet I'm happy to stick to Kali though (as long as it works still).
Is the chipcon tool available using just the pip install method? I don't remember seeing it, is that the swra one?

GravesJake, 2020-09-01:
No I don't believe chipcon is available in the pip release but it's definitely available in the latest nightly build. And yes, the swra functionality is exposed to the python chipcon API. I'm not too familiar with rpi compatible distros unfortunately, I run ubuntu for reference.

pickeditmate, 2020-09-01:
Thank you Jake, for all your help and assistance, I needed some serious guidance and you were there every step of the way. Thank you.

GravesJake, 2020-09-01:
No problem :)