greatscottgadgets/greatfet #350: Incomplete build instructions

pickeditmate, 2020-09-19:
It seems that the instructions are missing some steps, I have installed the latest Ubuntu desktop and failing at the step of sudo apt-get
Specifically these 5 packages are not found:
gcc-arm-none-eabi
libnewlib-arm-none-eabi
dfu-util
python-setuptools
python-yaml

Is there a specific version of Ubuntu that comes with these packages? I would like to be using python3 where possible as I know that's the direction you guys are deving for.

GravesJake, 2020-09-22:
What kind of not found error do you get? All of these packages show up in my installed apt packages in Ubuntu 20.04 (were not included with Ubuntu).

pickeditmate, 2020-09-23:
Ah I'm told I have to enable multiverse, sorry for the huge noob error on my part. PICNIC on my part.

closed 2020-09-23 by pickeditmate