greatscottgadgets/greatfet #351: Chipcon ID 65535

pickeditmate, 2020-09-20:
I am testing out the newest addition of greatfet chipcon CLI tool. 
When issuing a gf chipcon --chip-id I'm getting a readout of "Chip ID : 65535" which seems rather strange, so wondering if someone else who has an imme or yardstick can test theirs to see if they get the same. 

My wiring is J2.37 to !rst, 
J2.38 to DD, 
J2.39 to DC, 
power and ground to +2.5V and gnd respectively.

GravesJake, 2020-09-22:
Testing on yardstick I get `chip_id = 4355`. Make sure to call `chipcon.debug_init()` before doing anything else.

GravesJake, 2020-09-22:
My setup: 
J1.1 -> GND
J1.2 -> 3v3
J1.37 -> DD
J1.39 -> DC
J1.40 -> RESET

pickeditmate, 2020-09-23:
Thank you Jake, chip ID is now 259. I was using the pin mappings for the begonia neighbour https://github.com/greatfet-hardware/begonia directly. 

Thanks for the help and for the links too as I struggle finding which pins to use other than what's on the cheat sheet picture and silk screen. 

I wonder whether it's worthwhile having the pins as a comment in the swra library for others to not make the same mistake?

closed 2020-09-23 by pickeditmate