greatscottgadgets/greatfet #357: USB error when running "gf info" in a VM

potto216, 2020-11-09:
I receive a USB error when I run `gf info` which causes an exception and the code to end when running in a VM.  The error is usb.core.USBError: [Errno 5] Input/Output Error.

I am running the software in a VMWare VM using Ubuntu 18-4-4 and I think this is related to
a similar VMWare issue https://github.com/greatscottgadgets/greatfet/issues/250

I verified that the GreatFET is visible by the VM with `lsusb`. The output of the command is in lsusb_greatfet.txt.

The full output of the error when running `gf info` is in gf_info_output.txt.
This error occurs on
```
 File "/home/user/anaconda3/lib/python3.7/site-packages/pygreat/comms_backends/usb.py", line 354, in execute_raw_command
    self.LIBGREAT_REQUEST_NUMBER, self.LIBGREAT_VALUE_EXECUTE, flags, to_send, timeout)

351                     self.device.ctrl_transfer(
352                         usb.ENDPOINT_OUT | usb.TYPE_VENDOR | usb.RECIP_ENDPOINT,
353                         self.LIBGREAT_REQUEST_NUMBER, self.LIBGREAT_VALUE_EXECUTE, flags, to_send, timeout)
354
```

The full output of the error when additional debug information is selected (`LIBUSB_DEBUG=4 and USB_DEBUG=debug`) is in the file: gf_info_output_debug_verbose.txt.

In the above output file I had also modified "pygreat/comms_backends/usb.py" to print out additional information around line 354. The statements I added were:
```python
print("****************start ctrl_transfer************",flush=True);
print(f"flags = {flags}, to_send = {to_send}, timeout = {timeout}",flush=True);
print("****************end ctrl_transfer************",flush=True);
```

Their output is visible in "gf_info_output_debug_verbose.txt"

I researched the error and found a similar issue also posted on the pyusb repo (https://github.com/pyusb/pyusb/issues/216). They felt it was a VMWare issue however I could not find details supporting that. I use VMWare for the HackRF, RTL, LimeSDR, and AirSpy and have never had a problem with their USB pass though. If it is a problem with VMWare I would like to get additional details so I can work with VMWare to fix the issue.

The USB error has occurred with Python 3.6, 3.7, and 3.8 both using Anaconda installs and without. 

On the same computer that runs VMWare I was able to connect to the GreatFET One on the native Windows 10 install and also update the firmware. The firmware update did not change the error message.

I was wondering what I should do next for troubleshooting? 

I'm not sure if this is related, but I did find this issue interesting (https://github.com/pyusb/pyusb/issues/126) because it dealt with different behavior from ctrl_transfer with Linux and Windows backends. Since I'm running a VM Linux on a Windows 10 machine I didn't know if this could have any influence.

Thanks again for the hardware, software, and talks your company has provided.
[lsusb_greatfet.txt](https://github.com/greatscottgadgets/greatfet/files/5507501/lsusb_greatfet.txt)
[gf_info_output.txt](https://github.com/greatscottgadgets/greatfet/files/5507502/gf_info_output.txt)
[gf_info_output_debug_verbose.txt](https://github.com/greatscottgadgets/greatfet/files/5507503/gf_info_output_debug_verbose.txt)

potto216, 2020-12-17:
Just seeing if there are any suggestions on what I should do to further debug the issue?

potto216, 2021-02-15:
Let me know if there is any way I can help with this.  Thanks for looking at it!

patois, 2021-11-05:
I've had partial success with a Linux VM running inside VMWare Workstation by editing its .vmx configuration file. Shutting down the VM, then adding the following line and booting the VM again will make GreatFET devices recognizable but still not fully functional (there will be either exceptions or misbehavior depending on the gf script you're running).

`usb.generic.skipsetconfig = "TRUE"`

It may be worth trying/adding other .vmx configuration settings, some can be found on [sanbarrow.com](https://sanbarrow.com/vmx.html), others on [this page](https://s3.amazonaws.com/virtuallyghetto-download/hidden_vmx_params.html). Good luck!

potto216, 2021-11-24:
Thanks, I'll try that out.

robvinson, 2022-04-29:
Adding this line to my VMWare VMs .vmx file got the greatfet working for me:

```
usb.quirks.device0 = "0x1d50:0x60e6 allow, skip-reset, skip-refresh, skip-setconfig"
```

potto216, 2022-05-15:
Thanks so much @robvinson !!!!! This worked for me also. I'm going to close the issue because this is a great solution.

closed 2022-05-15 by potto216

meeas, 2022-06-27:
Awesome work @robvinson !!!  That has plagued my Black Hat course for the last few years, forcing everyone on Windows to use VirtualBox.  The VMware works just fine without modification on Mac and Linux, just not Windows.  And all other virtualization software on Windows works fine, just not VMware................until now.  :-)