greatscottgadgets/greatfet #359: Chipcon error on read and write.

DanielSchetritt, 2020-11-27:
# Summary
I am trying to backup the software of my IM-ME, and then write new software to it.
Running `greatfet chipcon --chip-id` gives me `259`.
But it fails on `greatfet chipcon --read ~/im-me-firmware-backup.hex --length 0x8D98`. Attempting to write data to it results in the same error.
# Hardware Setup
```
J1.01 -> GND
J1.02 -> 3v3
J1.37 -> DD
J1.39 -> DC
J1.40 -> RESET
```
# Software Setup
This assumes that `udev` configurations are set and that you have the `plugdev` group. In order to get the chipcom subcommand, I had to install from source by following the instructions on [PyPI](https://pypi.org/project/GreatFET/)  thus:
```bash
pip3 uninstall greatfet pygreat
git clone https://github.com/greatscottgadgets/greatfet
cd greatfet
git submodule init && git submodule update
# Install libgreat from source
cd libgreat/host && python3 setup.py build && sudo python3 setup.py install && cd ../..
# Install greatfet from source
cd host && python3 setup.py build && sudo python3 setup.py install
```
# Error Trace
This is the error I am getting when running `greatfet chipcon --read ~/im-me-firmware-backup.hex --length 36248`
```
➜  ~ greatfet chipcon --read ~/im-me-firmware-backup.hex --length 36248
Reading 36248 bytes starting at address 00...
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 755, in execute_command
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 565, in pack
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 1176, in int_array_arguments
struct.error: required argument is not an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/greatfet_chipcon", line 11, in <module>
    load_entry_point('GreatFET===b-v2020.1.2-.dev-git.b-72066c3c-', 'console_scripts', 'greatfet_chipcon')()
  File "/usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_72066c3c_-py3.8.egg/greatfet/commands/greatfet_chipcon.py", line 62, in main
    read_flash(chipcon, args.read, args.address, args.length, log_function)
  File "/usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_72066c3c_-py3.8.egg/greatfet/commands/greatfet_chipcon.py", line 77, in read_flash
    data = programmer.read_flash(start_address=start_address, length=length)
  File "/usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_72066c3c_-py3.8.egg/greatfet/programmers/chipcon.py", line 288, in read_flash
    flash_data.extend(self.read_flash_page(i))
  File "/usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_72066c3c_-py3.8.egg/greatfet/programmers/chipcon.py", line 276, in read_flash_page
    return self.read_code_memory(linear_address & 0xFFFF, FLASH_PAGE_SIZE)
  File "/usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_72066c3c_-py3.8.egg/greatfet/programmers/chipcon.py", line 116, in read_code_memory
    self.run_instruction(0x75, 0xc7, (bank * 16) + 1)               # MOV MEMCTR, (bank * 16) + 1
  File "/usr/local/lib/python3.8/dist-packages/GreatFET-b_v2020.1.2_.dev_git.b_72066c3c_-py3.8.egg/greatfet/programmers/chipcon.py", line 98, in run_instruction
    return self.api.debug_instr(instruction)
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 1108, in method
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 1263, in execute_command
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 762, in execute_command
  File "/home/daniel/.local/lib/python3.8/site-packages/future/utils/__init__.py", line 446, in raise_with_traceback
    raise exc.with_traceback(traceback)
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 755, in execute_command
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 565, in pack
  File "/usr/local/lib/python3.8/dist-packages/pygreat-b_v2020.1.2_.dev_git.b_e8351008_-py3.8.egg/pygreat/comms.py", line 1176, in int_array_arguments
struct.error: invalid arguments in call to RPC `debug_instr`; innner message: required argument is not an integer; format: <*B
```

DanielSchetritt, 2020-11-27:
I just noticed a `Makefile` in this repository. I am going to try installing greatfet using the makefile like this:
(I'm on Pop OS; an Ubuntu derivative)
```bash
sudo apt install gcc-arm-none-eabi libnewlib-arm-none-eabi dfu-util python-setuptools python-yaml
git clone https://github.com/greatscottgadgets/greatfet
cd greatfet/
make
```

blysmth, 2020-12-23:
Hi @DanielSchetritt I have the exact same problem, using Linux Mint 20; same error messages.

blysmth, 2020-12-23:
I figure it out, you need to update the firmware with the current version before executing commands.

Still, I tried writing Mike's Spectrum Analyzer hex to my IM-ME and now it's dead :(

Qyriad, 2021-01-06:
I'm looking into this now. The verification is failing on my end, so something is definitely not quite right.

Qyriad, 2021-01-06:
@DanielSchetritt @blysmth so it turns out the reason it was failing for me was a totally unrelated environment issue. Why it's not working for *you*, however, is that you're passing an Intel hex (`.hex`) file. The chipcon programmer writes whatever data you pass it literally, as binary. However, once #362 is merged, the `greatfet chipcon` command will auto-detect Intel hex files and convert them to binary for you, so everything should work as expected after that.

blysmth, 2021-01-06:
Thank you @Qyriad, to clarify: does that mean the hex I downloaded from http://ossmann.blogspot.com/2010/03/16-pocket-spectrum-analyzer.html was't properly flashed?

Also, is there a chance of reviving my IM-ME once #362 is merged?

Qyriad, 2021-01-06:
That means the hex you downloaded was flashed without converting it to binary first. As for reviving your IM-ME, so long as you can still run something like `greatfet chipcon --chip-id`, and the firmware you're trying to flash is fine, then you should be able to revive it by reflashing it with `greatfet chipcon -w path_to_hex_file`.

blysmth, 2021-01-06:
Thank you so much. I feel like such a silly rookie, so out of my depth! 😬

Qyriad, 2021-01-06:
No worries! My apologies for the unclear semantics. On the plus side, #362 is merged now. Before trying to re-flash, make sure you install the updated version of the greatfet host stack with something like:
```sh
$ cd greatfet
$ git pull origin master
$ pip3 install --user --upgrade ./host
```

blysmth, 2021-01-06:
Works perfectly, thanks again @Qyriad. You're a star.
[image]

Qyriad, 2021-01-06:
Awesome. I'm going close this now; glad everything is working :)

closed 2021-01-06 by Qyriad