greatscottgadgets/greatfet #363: autodetect not working on OpenBSD

zuernerd, 2021-01-10:
Hey im trying to use the GreatFETOne on OpenBSD but there are some issues.

When running
```
>>> gf = greatfet.GreatFETBoard.autodetect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/pygreat-b_v2020.1.2_.dev_git.b_78e91fca_.dirty-py3.8.egg/pygreat/board.py", line 83, in autodetect
pygreat.errors.DeviceNotFoundError
```
The full output of the error when additional debug information is selected `LIBUSB_DEBUG=4` is in the file:
[autodetect_out.log](https://github.com/greatscottgadgets/greatfet/files/5792807/autodetect_out.log)

If i understand correctly the autodetect method will create new instances for every subclass. It will create several `USBCommsBackend`objects. During initialization they will all run `usb.core.find(**device_identifiers))` and `self.device.get_active_configuration():` on the same device. 
This results in several `libusb_open` executions on the same `ugen(4)` device, which fail with EBUSY.

Any ideas on this issue?

A workaround is to run instead:
```
>>> gf = greatfet.boards.one.GreatFETOne()
```
but then the command utilities still don't work. Is there actually a way to use this workaround for FaceDancer?

ktemkin, 2021-01-11:
Thanks for the report, and for the detailed logging.

> Any ideas on this issue?

Unfortunately, not a lot of testing has been done on OpenBSD; so I suspect there are platform quirks that are yet to be written. Here, it looks like we might want to back off and wait on `EBUSY`.

> Is there actually a way to use this workaround for FaceDancer? 

How are you using FaceDancer? The workaround will be different depending on whether you're using the new FaceDancer syntax (e.g. are you importing from `facedancer.future`, currently?)

zuernerd, 2021-01-16:
> Unfortunately, not a lot of testing has been done on OpenBSD; so I suspect there are platform quirks that are > yet to be written. Here, it looks like we might want to back off and wait on `EBUSY`.

I think waiting on `EBUSY` wont solve the problem. I it is caused by the subclasses who will run `usb.core.find()`.
Therefore we will end up with several objects for the same device. Then when we want to open the device we will get `EBUSY`. The following example code will work on Linux (raspian) but will fail on OpenBSD as shown here:

```
>>> import usb
>>> dev1 = usb.core.find(idVendor=0x1d50)
>>> dev2 = usb.core.find(idVendor=0x1d50)
>>> dev1.get_active_configuration()
<CONFIGURATION 1: 500 mA>
>>> dev2.get_active_configuration()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/usb/core.py", line 894, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "/usr/local/lib/python3.8/site-packages/usb/core.py", line 98, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/usb/core.py", line 232, in get_active_configuration
    self.managed_open()
  File "/usr/local/lib/python3.8/site-packages/usb/core.py", line 98, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/usb/core.py", line 116, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 789, in open_device
    return _DeviceHandle(dev)
  File "/usr/local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 637, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 589, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
```

> How are you using FaceDancer? The workaround will be different depending on whether you're using the new FaceDancer syntax (e.g. are you importing from `facedancer.future`, currently?)

Im fairly new to the project and would like to play around with it. Therefore i guess it would be best to stick to the new syntax ?

Thank you for your support so far

mossmann, 2021-11-05:
Based on your last trace, this appears to be a pyusb and/or libusb bug on your platform. If you want to pursue a solution, I think you'll need to look upstream.

closed 2021-11-05 by mossmann