greatscottgadgets/greatfet #364: AttributeError: 'array.array' object has no attribute 'tostring' - Python 3.9

godenaM, 2021-02-10:
gf fw --auto                           
Trying to find a GreatFET device...
Traceback (most recent call last):
  File "/home/kali/.local/bin/greatfet_fw", line 8, in <module>
    sys.exit(main())
  File "/home/kali/.local/lib/python3.9/site-packages/greatfet/commands/greatfet_firmware.py", line 200, in main
    device = parser.find_specified_device()
  File "/home/kali/.local/lib/python3.9/site-packages/greatfet/utils.py", line 219, in find_specified_device
    device = self._find_greatfet(args)
  File "/home/kali/.local/lib/python3.9/site-packages/greatfet/utils.py", line 312, in _find_greatfet
    return GreatFET(serial_number=args.serial)
  File "/home/kali/.local/lib/python3.9/site-packages/greatfet/greatfet.py", line 33, in GreatFET
    return GreatFETBoard.autodetect(**board_identifiers)
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/board.py", line 72, in autodetect
    if subclass.accepts_connected_device(**device_identifiers):
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/board.py", line 168, in accepts_connected_device
    potential_device = cls(**device_identifiers)
  File "/home/kali/.local/lib/python3.9/site-packages/greatfet/board.py", line 86, in __init__
    super(GreatFETBoard, self).__init__(*args, **kwargs)
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/board.py", line 245, in __init__
    if self.HANDLED_BOARD_IDS and (self.board_id() not in self.HANDLED_BOARD_IDS):
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/board.py", line 274, in board_id
    return self.apis.core.read_board_id()
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/comms.py", line 1108, in method
    return self.execute_command(verb_number, in_format, out_format, name=name, class_name=class_name,
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/comms.py", line 1263, in execute_command
    return self.comms_backend.execute_command(self.CLASS_NUMBER, verb, in_format,
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/comms.py", line 770, in execute_command
    raw_result = self.execute_raw_command(class_number, verb, payload, timeout,
  File "/home/kali/.local/lib/python3.9/site-packages/pygreat/comms_backends/usb.py", line 377, in execute_raw_command
    return response.tostring()
AttributeError: 'array.array' object has no attribute 'tostring'

I'm having this issue on Win10 and KaliLinux. Is there any solution for it?

tarikku, 2021-02-10:
Hi,
This was recently reported and fixed as it seems, see here: https://github.com/greatscottgadgets/greatfet/issues/360

markoHUN, 2021-02-11:
Thank you! Now I fixed this issue based on your link,  but I get timeout error. Do you know what could cause it?

closed 2021-02-12 by markoHUN