greatscottgadgets/greatfet #366: Does greatfet support umap2 tool?

shtry, 2021-02-17:
Hi.

Does greatfettool support  umap2?

umap2 link : https://github.com/nccgroup/umap2

In my opinion, it seems to be an error caused by not finding the physical device. "/dev/ttyUSB0".

I get a device error as below.

```
$ sudo umap2emulate -P fd:/dev/ttyUSB0 -C fidi
[INFO  ] Loading physical interface: fd:/dev/ttyUSB0
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/umap2emulate", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/umap2/apps/emulate.py", line 53, in main
    app.run()
  File "/usr/local/lib/python3.8/dist-packages/umap2/apps/emulate.py", line 35, in run
    self.phy = self.load_phy(self.options['--phy'])
  File "/usr/local/lib/python3.8/dist-packages/umap2/apps/base.py", line 66, in load_phy
    s = Serial(dev_name, 115200, parity=PARITY_NONE, timeout=2)
  File "/usr/local/lib/python3.8/dist-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/usr/local/lib/python3.8/dist-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
```

shtry, 2021-02-17:
Is there a separate USB device mode?

https://github.com/greatfet-hardware/azalea/blob/master/README.md

```
power switching for use of USB1 as either host or device
```

straithe, 2022-08-10:
Do you have a link to the issue you open on umap2?

straithe, 2023-11-12:
@shtry I didn't ever hear back from you. Do you still need assistance with this issue?

dhingos, 2024-06-26:
Hello, I realize this is an old thread, but I came across this issue seeking a similar solution. If it has not been resolved, I would like to reopen this and try to find a solution.

antoinevg, 2024-06-27:
I'm uncertain of the state of umap2 support for GreatFET but the problem may be that the operating system is not recognizing the GreatFET or assigning it a different port than `/dev/ttyUSB0`

What output do you see when you enter:

```sh
greatfet info -a
```

Also, try unplugging the device and run the following in a terminal:

```sh
sudo dmesg -w
```

What, if any, log entries come up when you plug the device in?

Finally, unplug the device and type the following, then re-plug the device and type the following again:

```sh
ls /dev
```

Are there any new entries after you plugged the device back in? If so, that is probably the port the os assigned to GreatFET.

dhingos, 2024-06-27:
> I'm uncertain of the state of umap2 support for GreatFET but the problem may be that the operating system is not recognizing the GreatFET or assigning it a different port than `/dev/ttyUSB0`
> 
The issue is that the GreatFET, while somewhat billed by many as a replacement for Facedancer, does not get a tty port. If you search for "umap2 greatfet" you will find several articles and other places across the web (including Github) where people mention the use of umap2 with GreatFET, but experience seems to show that they do not work together. If I understand correctly the GreatFET acts as a UART device and the "gf" toolset allow interaction with the GreatFET using this mode of communication. I have read that nu-map supports it but have not seen this in practice either. I am still digging into this. 

I will get the output of what you ask as soon as I have the chance today but I figured I'd mention this anyway.

martinling, 2024-06-27:
> If I understand correctly the GreatFET acts as a UART device and the "gf" toolset allow interaction with the GreatFET using this mode of communication.

No, the GreatFET doesn't show up as a serial port at all. The `gf` tools control it over a custom USB protocol that is defined in [libgreat](https://github.com/greatscottgadgets/libgreat/). The [Facedancer](https://github.com/greatscottgadgets/facedancer/) library has a `greatdancer` backend which talks the libgreat USB protocol, as well as a `goodfet` backend which supports the serial protocol used by the old GoodFET-based boards like Facedancer21.

But umap2 doesn't go through the Facedancer API, it has its own backends: `facedancer`, which supports the serial GoodFET-based boards, `gadgetfs`, and `raspdancer`. There's no backend for libgreat/GreatFET, so umap2 isn't usable with GreatFET One.

There is a fork of umap2, [nu-map](https://github.com/usb-tools/nu-map) which removes all umap2's internal backends and goes through the Facedancer (2.x) API instead. So in theory, that version should work with a GreatFET One, but I don't know if it's currently in a working state.

dhingos, 2024-06-27:
A look over at the nu-map issues 

> No, the GreatFET doesn't show up as a serial port at all. The `gf` tools control it over a custom USB protocol that is defined in [libgreat](https://github.com/greatscottgadgets/libgreat/). The [...]

First, thanks for your insight. Second, a quick look at numap shows similar commands to umap and umap2, such that numap's device emulation, device support scanning and fuzzing require specifying a target tty device as an argument on the command line. The hardware section of numap's readme does not list GreatFET as a supported board, I'm assuming for the same reasons as the other varieties of umap. It seems the path of least resistance here may be to attempt to port desired functionality into the latest GreatFET compatible facedancer implementation.

martinling, 2024-06-27:
> Second, a quick look at numap shows similar commands to umap and umap2, such that numap's device emulation, device support scanning and fuzzing require specifying a target tty device as an argument on the command line.

In nu-map the `-P` argument is still required by the commands but what you put there doesn't matter. Internally that argument gets passed to `NumapApp.load_phy` which ignores it and just calls the `FacedancerUSBApp` constructor:

https://github.com/usb-tools/nu-map/blob/master/numap/apps/base.py#L58-L60

> The hardware section of numap's readme does not list GreatFET as a supported board, I'm assuming for the same reasons as the other varieties of umap.

I think that's just because it's a very quick and dirty fork that never got cleaned up and finished. In principle it looks like it might work, at least partially.

dhingos, 2024-06-30:
I will give it a shot and report back. Thanks for responding!

dhingos, 2024-07-22:
Sorry this took me so long...I tried a number of combinations of commands, I thought that something like: 

numap-scan -P=/dev/greatfet-one1-6 -v

would work but I just get more errors, such as:

  File "/usr/local/lib/python3.10/dist-packages/numap/core/usb_device.py", line 13, in <module>
    from facedancer.USBDevice import USBDevice as BaseUSBDevice
ModuleNotFoundError: No module named 'facedancer.USBDevice'

I think I will go over to the numap project and discuss there, it is getting clearer that this is not really an issue with greatfet.

martinling, 2024-07-22:
What version of the `facedancer` library are you using? It would have been written to use the Facedancer 2.x API, so you'll need to use the 2.9.0 release to get anywhere I think. Either that or update it to use the new 3.0 API.

dhingos, 2024-07-22:
Aah great catch. Somehow through all my troubleshooting I wound up replacing 2.9 with 3.0. Backed it out and reinstalled. That error is gone now, but still no dice. I will keep at it though, it seems like this is so close....I posted on the numap project as well.