greatscottgadgets/greatfet #367: greatfet_logic broken when using num channels != 1, 8

docprofsky, 2021-02-18:
Using greatfet_logic with the number of channels (-n) not equal to 8 or 1 results in the below error.
`[Errno 5] Input/Output Error`

I have done testing and I believe this error was introduced by e3f25fd2c4461c3de64d6b7f4e371b0001168eda.

Below is the dmesg log from the GreatFET after running `greatfet_logic -n 5 -f 10e6 -o /dev/null --debug`
```
[------------] System started after hard reset / power cycle.
[      117740] System clock bringup complete.
[      117773] Configuring board pins...
[      117821] GreatFET initialization complete!
[     1997245] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[     1997856] sgpio: warning: cannot create a %hhu-bit bus; creating an 8-bit bus instead.
[     1998183] --- SGPIO state dump (1 functions) --- 
[     1998197] ======================== 
[     1998208] 
[     1998215] Software function configuration: 
[     1998227]     function 0: STREAM IN   io slice: A /  0  buffer order/size: 15/32768  buffer position: 0
[     1998263] 
[     1998270] Pin configuration: 
[     1998280]     pin usage mask: 0x001f
[     1998292]     GPIO output enable: 0000
[     1998305]     SGPIO 0:  INPUT                        
[     1998325]     SGPIO 1:  INPUT                        
[     1998345]     SGPIO 2:  INPUT                        
[     1998365]     SGPIO 3:  INPUT                        
[     1998385]     SGPIO 4:  INPUT                        
[     1998406] 
[     1998413] Slice configuration: 
[     1998424]     slice usage mask: 0x0f35
[     1998436]     slice[ 0] / A: 8-bit parallel    input: external pin             clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts  [IRQ]
[     1998486]     slice[ 2] / C: 8-bit parallel    input: slice J (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998540]     slice[ 4] / E: 8-bit parallel    input: slice I (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998592]     slice[ 5] / F: 8-bit parallel    input: slice K (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998647]     slice[ 8] / I: 8-bit parallel    input: slice A (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998698]     slice[ 9] / J: 8-bit parallel    input: slice E (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998751]     slice[10] / K: 8-bit parallel    input: slice C (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998805]     slice[11] / L: 8-bit parallel    input: slice F (chain 8 deep)   clock: counter, div:  19/ 20   shifts: always   data/shadow swap every  32 shifts 
[     1998861] 
[     1998868] Slice contents: 
[     1998877]     slice[ 0] / A:     data: 00000000    shadow: 00000000
[     1998901]     slice[ 2] / C:     data: 00000000    shadow: 00000000
[     1998925]     slice[ 4] / E:     data: 00000000    shadow: 00000000
[     1998949]     slice[ 5] / F:     data: 00000000    shadow: 00000000
[     1998974]     slice[ 8] / I:     data: 00000000    shadow: 00000000
[     1998998]     slice[ 9] / J:     data: 00000000    shadow: 00000000
[     1999022]     slice[10] / K:     data: 00000000    shadow: 00000000
[     1999046]     slice[11] / L:     data: 00000000    shadow: 00000000
[    12866005] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000

```

olistrik, 2025-10-23:
Was this ever resolved?  I can't seem to sample anything over 15.6MHz with any number of channels. With numbers other than 1 & 8, I can sample up to around 1MHz, but it's really hit and miss.

olistrik, 2025-10-23:
What I find really odd is PulseView can get up to 102MHz with 2 probes; seemingly indefinitely. While Sigrok will only go up to 40.8MHz. I thought PulseView was just a GUI for Sigrok, so why they don't perform to the same level I don't know.