greatscottgadgets/greatfet #370: Python UART baud rate bug

tillostermann, 2021-04-09:
Hi,

not sure if anyone here can help out, but I'll give it a shot. I'm an hardware engineer and like the possibility to directly use the typical standard low level interfaces like I2C, SPI or UART directly with Python. For development I'm using Spyder 4 on Ubuntu, installed gf via pip. 
So far things are great, but I'd like to adjust the UART baud rate so that I can use it in my application. And it turns out, that it does not work, the set baud rate arbitrarily changes real baud rate, checked with a Saleae logic analyzer. The standard 115200 baud rate works fine.

_from greatfet import GreatFET
import time
gf = GreatFET()
gf.uart.update_parameters(baud=19200, data_bits=8, stop_bits=1,parity=None )   
time.sleep(0.1)
gf.uart.write(b"hi")_

Using the debug function drops me into _greatfet/interfaces/uart.py_ and _pygreat/comms.py_ . Here's were I get stuck... Looking at the behavior it appears that it might be a data type issue? Did anyone else see similar issues?

BR
Till Ostermann

@Qyriad

miek, 2021-04-30:
I haven't been able to reproduce this so far. When I run your script, the output looks correct to me:
[image]

Is it possible that you're running an older version? You might be hitting #328.
Could you post the output of `gf info -H`?

tillostermann, 2021-05-01:
Was on:
[image]

Updated to:
[image]

Did indeed solve the issue! @miek thanks a lot for your answer!

closed 2021-05-01 by tillostermann