greatscottgadgets/greatfet #371: Error flashing older GreatFET ("28 January 2017" silkscreen)

alvarop, 2021-04-16:
Trying to update the firmware on and old-ish greatfet and getting the following error:
```
greatfet fw --auto
Trying to find a GreatFET device...
GreatFET [legacy FW] found. (Serial number: 0000000000000000e666c4d113207436)
Writing data to SPI flash...

Traceback (most recent call last):                                              
  File "/home/alvaro/.local/lib/python3.8/site-packages/greatfet/programmers/firmware.py", line 76, in write
    self.comms.get_exclusive_access()
AttributeError: 'LegacyCommsAdapter' object has no attribute 'get_exclusive_access'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/alvaro/.local/bin/greatfet_fw", line 8, in <module>
    sys.exit(main())
  File "/home/alvaro/.local/lib/python3.8/site-packages/greatfet/commands/greatfet_firmware.py", line 214, in main
    spi_flash_write(device, args.write, args.address, log_function)
  File "/home/alvaro/.local/lib/python3.8/site-packages/greatfet/commands/greatfet_firmware.py", line 56, in spi_flash_write
    device.onboard_flash.upload(filename, address, erase_first=True,
  File "/home/alvaro/.local/lib/python3.8/site-packages/greatfet/programmers/firmware.py", line 280, in upload
    self.write(data, address, erase_first, progress_callback)
  File "/home/alvaro/.local/lib/python3.8/site-packages/greatfet/programmers/firmware.py", line 79, in write
    self.comms.release_exclusive_access()
AttributeError: 'LegacyCommsAdapter' object has no attribute 'release_exclusive_access'
```

This is what I get with the info command:
```
greatfet info      
Found a GreatFET [legacy FW]!
  Board ID: 0
  Firmware version: git-e3d6e4e
  Part ID: 300a00a069436e
  Serial number: 0000000000000000e666c4d113207436

  !!! WARNING !!!
    This device's firmware is too out of date! It must be upgraded before
    it can be used. See https://github.com/greatscottgadgets/greatfet/wiki
    for more information.
```

Running ubuntu 20.04 with python3 and the udev rules installed
Board is labeled "28 January 2017"

alvarop, 2021-04-16:
Using the DFU recovery procedure as recommended here worked!
https://github.com/greatscottgadgets/greatfet/tree/master/firmware#dfu-mode-running-from-ram

```
greatfet info            
Found a GreatFET One!
  Board ID: 0
  Firmware version: v2021.2.1
  Part ID: a0000a305e4765
  Serial number: 0000d1c466e637973013
```

straithe, 2021-05-04:
Thank you for the update, @alvarop!

closed 2021-05-04 by straithe