greatscottgadgets/greatfet #373: USB Mass Storage Device not working

mjargay, 2021-05-06:
Hello,

My code appears to hang on the SCSI READ (0x28) and WRITE (0x2A) when the amount of data exceeds 512. The code appears to hang at greatdancer.py line 250. The greatdancer doesn't appear to ever receive a transfer is complete. I am able to verify that all data is sent from the HOST to the DEVICE by sniffing the traffic using Wireshark during a WRITE (0x2A). However the code only appears to receive a packet each time I kill the app and restart it. The READ (0x28) completely hangs the USB bus until I unplug the device.

Thanks

loneicewolf, 2021-05-07:
Can you send a screenshot? I had this problem a long (**awhile!**) ago,  and I (Don't remember Exactly how I) solved it, but can you also - (if possible) provide the code you're trying to run?

Regards

Will

mjargay, 2021-05-07:
I am using the facedancer library.

The issue happens even when i use facedancer-umass.py in the facedancer repository. It seems that the get_status(self.GET_ENDPTSETUPSTAT) from this repository isn't updating the status to say that the board can take or receive new traffic. There is a lot of code but its the old code ported to FaceDancer 3.0. The legacy code has the issue as well.

Thanks

loneicewolf, 2021-05-08:
Thanks for pointing it out, will have a look into it.

Will

robvinson, 2022-04-29:
It appears that the 2021.2.1 firmware may have introduced some issues that didn't previously exist.

I couldn't get the facedancer `legacy-applets/facedancer-umass.py`  script to successfully present a disk image to a target system as a storage device. The script would always hang after the target Windows host attempted SCSI WRITE(10) operation. After downgrading greatfet to 2020.1.2 and adjusted the firmware back to 2020.1.2 the script started working just fine.  Perhaps one of the few changes in greatdancer.c broke something? 

Diff between the versions: https://github.com/greatscottgadgets/greatfet/compare/v2020.1.2..v2021.2.1#diff-74f72ddc92cff5c0fe5f5cf8edcf070ae01aca08b3c830263828106976ca0d90

For posterity, to downgrade:
```
pip uninstall greatfet
pip install greatfet==2020.1.2
gf fw --auto
```

mjargay, 2022-08-07:
Thanks robvinson for figuring that out. I logged in today to see if the issue was fixed yet and only saw your comment. The project looks like it may be abandoned but here is the fix if you are still using the hardware.

Changing "typedef char packet_buffer[4096];" to "typedef char packet_buffer[512];" in firmware/greatfet_usb/classes/greatdancer.c line #29 fixed the issue. I am assuming this needs to be the same size as the one in firmware/greatfet_usb/legacy_apis/usb_api_usbhost.c. I tried changing usb_api_usbhost.c to 4096 instead with no luck. So may be a buffer fill issue instead.

I can now finally use the device for what I purchased it for.

Thanks again!

straithe, 2022-08-19:
I'm closing this issue as it seems to have been resolved. Thank you @robvinson for your contributions.

PR #425 (2024-01-18): greatdancer: fix facedancer usb mass storage example not working
This commit reverts 516d98a and fixes #373.
closed 2024-01-18 by antoinevg