greatscottgadgets/greatfet #383: Update UART interface to enable accessing UART1 and UART2

aussieklutz, 2021-07-09:
The Azalea board appears to have two additional UART interfaces, as indicated as U1 and U2 in the table at: https://github.com/greatfet-hardware/azalea#j1-neighbor1-pinout
The UART interface is currently initiated thusly:

`    def __init__(self, board, baud=115200, data_bits=8, stop_bits=1, parity=None, uart_number=0):
        """
        Args:
            board -- GreatFET board whose UART lines are to be controlled
        """

        self.api = board.apis.uart

        # We'll only initialize the on-board UART on demand; so we can have a UART object
        # around by default, without necessarily having it
        self.initialized = False

        # Store our UART control parameters.
        self.baud        = baud
        self.data_bits   = data_bits
        self.stop_bits   = stop_bits
        self.parity      = parity or self.PARITY_NONE
        self.uart_number = 0
        self.actual_baud = 0`

To the best I can follow, an additional UART interface can be instantiated in this way:

`gf._add_interface("uart1", UART)`

The __init__ for the UART interface does not reference the provided uart_number, hence not allowing access to the additional UART interfaces.

aussieklutz, 2021-07-09:
Further to this, it looks like the additional UART1 and UART2 pins are not defined in: https://github.com/greatscottgadgets/greatfet/blob/master/firmware/boards/azalea/pins.h

aussieklutz, 2021-07-09:
Looking in libgreat, the LCP4330 does not have the pins defined for the additional UARTS:
https://github.com/greatscottgadgets/libgreat/blob/master/firmware/platform/lpc43xx/drivers/platform_uart.c

`/**
 * Mapping that provides the location of the default pins for each UART.
 */
uart_pins_t uart_pins[] = {
	{
		.tx = { .group = 9, .pin = 5, .function = 7 },
		.rx = { .group = 9, .pin = 6, .function = 7 }

		// TODO: add flow control pins!
	},

	// FIXME: add the other UARTS!
};
`

aussieklutz, 2021-07-09:
Made pull request to add the pin definitions.

aussieklutz, 2021-07-09:
Best I can tell, https://github.com/greatscottgadgets/greatfet/blob/master/firmware/boards/azalea/pins.h does not need to be updated as it has been superseded by the libgreat definitions.

Looks like the following are already written to support multiple UART:
https://github.com/greatscottgadgets/greatfet/blob/master/firmware/greatfet_usb/classes/uart.c
https://github.com/greatscottgadgets/libgreat/blob/master/firmware/drivers/uart/ns16550.c

grvvy, 2021-08-12:
Sorry for the delay, I'll look into patching this in. Thank you for your work!

grvvy, 2022-06-06:
Fixed here https://github.com/greatscottgadgets/libgreat/commit/55c89c1c6d3205e0c2e8a4ece99fa82ac71716b4

closed 2022-06-06 by grvvy