greatscottgadgets/greatfet #386: Ambiguous argument error when building host tools

cdwilson, 2021-08-07:
When I try to build `/libgreat/host/` or `/host/` per the [documentation](https://greatfet.readthedocs.io/en/latest/getting_started_firmware_development.html#fresh-install), I get a `fatal: ambiguous argument` error like the following:

```
$ python setup.py build
fatal: ambiguous argument 'bv2021.2.1': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
/Users/chris/.anyenv/envs/pyenv/versions/3.9.6/envs/greatfet-upstream/lib/python3.9/site-packages/setuptools/dist.py:498: UserWarning: The version specified ("b'v2021.2.1'.dev+git.b'9c57c249'") is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
...
```

I believe this ambiguous argument error can be fixed by ktemkin/even-better-setuptools-git-version#1 but this PR has been open for a couple years and has not been merged (I think https://pypi.org/project/even-better-setuptools-git-version/ will also need to be updated if the PR is merged).

For now, I'm using a branch in my greatfet fork that uses this PR, so this change can be easily tested by doing the following:
```
pip install -U pip setuptools wheel pyyaml
git clone --recursive --branch cdwilson/fix-even-better-setuptools-git-version https://github.com/cdwilson/greatfet.git

pushd libgreat/host/
python setup.py build
pip install -e .
popd

pushd host/
python setup.py build
pip install -e .
popd
```

grvvy, 2021-08-12:
I haven't seen this one before, have you tried using `python3` explicitly?

cdwilson, 2021-08-21:
@grvvy I'm seeing the same issue using `python3` explicitly.  

Can you try running the following and let me know if you get the same error?

```bash
pip install even-better-setuptools-git-version
cd greatfet/
python
```

```python
>>> from better_setuptools_git_version import *
>>> get_tag_commit_sha(get_tag())
fatal: ambiguous argument 'bv2021.2.1': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
b''
```

antoinevg, 2024-05-21:
We're no longer using `even-better-setuptools` so I'm going to close this.

closed 2024-05-21 by antoinevg